dlenski/tetherback #1: Corrupted gzip file which won't unzip

theduffmaster, 2016-03-20:
Device: Nexus 5
OS: Mac OSX
First off thanks for making the program. It seems to be working well except whenever I use nandroid.py the bigger files, such as the system.img.gz and the userdata.img.gz, fail when decompressing the gzip. The "crc fails" and it produces an img file about 1/5th the size of the img.gz file. However, the bootloader.img.gz decompresses fine. Is their anyway that you could disable the gzipping or do you know another way to fix the corrupted file? I've edited the file (line 46) to get rid of the gzip, but all it does is output an empty .img file.

dlenski, 2016-03-20:
Thanks, this is useful. I strongly suspect that this is an issue with your build of `adb` mangling the CR/LF characters :-( (http://stackoverflow.com/questions/13578416/read-binary-stdout-data-from-adb-shell/20141481#20141481)

What version of ADB are you using?

I will add an option to use TCP forwarding rather than piped output to transmit the data to the host, even though it makes the code messier. Maybe I can also figure out a way to do an md5sum on both the device and host sides to verify that there's no corruption.

I may have some time to work on it later tonight or tomorrow.

theduffmaster, 2016-03-21:
I'm using Android Debug Bridge version 1.0.32. Also, I did apply the stty fix from the stackoverflow answer, but it seems as if the gzip issue is still there after that.

dlenski, 2016-03-21:
Please grab the latest version and try this: `tetherback.py -t`

    -t: transfer files using TCP forwarding, rather than pipe
        (should avoid the image corruption issue)

theduffmaster, 2016-03-22:
It seems like it does fix the corruption issue! Here is the terminal output, created a separate issue for the error at the end of the terminal output.
```
`Device reports TWRP kernel (3.4.0-bricked-hammerhead-twrp-g7b77eb4).
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:03                                             
--  --------  ----------  ----------  ---------------------------
    NAME      DEVICE      SIZE (KiB)  BACKUP?
1   modem     mmcblk0p1   65536       (skip)
2   sbl1      mmcblk0p2   1024        (skip)
3   rpm       mmcblk0p3   512         (skip)
4   tz        mmcblk0p4   512         (skip)
5   sdi       mmcblk0p5   512         (skip)
6   aboot     mmcblk0p6   512         (skip)
7   pad       mmcblk0p7   2048        (skip)
8   sbl1b     mmcblk0p8   1024        (skip)
9   tzb       mmcblk0p9   512         (skip)
10  rpmb      mmcblk0p10  512         (skip)
11  abootb    mmcblk0p11  512         (skip)
12  modemst1  mmcblk0p12  3072        (skip)
13  modemst2  mmcblk0p13  3072        (skip)
14  metadata  mmcblk0p14  512         (skip)
15  misc      mmcblk0p15  16384       (skip)
16  persist   mmcblk0p16  16384       (skip)
17  imgdata   mmcblk0p17  3072        (skip)
18  laf       mmcblk0p18  22528       (skip)
19  boot      mmcblk0p19  22528       boot.img.gz (raw image)
20  recovery  mmcblk0p20  22528       (skip)
21  fsg       mmcblk0p21  3072        (skip)
22  fsc       mmcblk0p22  512         (skip)
23  ssd       mmcblk0p23  512         (skip)
24  DDR       mmcblk0p24  512         (skip)
25  system    mmcblk0p25  1048576     system.img.gz (raw image)
26  crypto    mmcblk0p26  30720       (skip)
27  cache     mmcblk0p27  716800      (skip)
28  userdata  mmcblk0p28  28551146    userdata.img.gz (raw image)
29  grow      mmcblk0p29  5           (skip)
--  --------  ----------  ----------  ---------------------------
Saving backup images in nandroid-backup-2016-03-21--18-48-09/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.img.gz: 100% Time: 0:00:11 844.82 kB/s                                   
Saving partition system (mmcblk0p25), 1024 MiB uncompressed...
  system.img.gz: 100% Time: 0:10:17 944.49 kB/s                                 
Saving partition userdata (mmcblk0p28), 27881 MiB uncompressed...
  userdata.img.gz: 100% Time: 3:48:44   1.69 MB/s                                                                                                                                                                                             
error: cannot remove listener
error: cannot remove listener
Traceback (most recent call last):
  File "/Volumes/GLYPH-2 TB/nexus5recovery/tetherback.py", line 125, in <module>
    sp.check_call(('adb','forward','--remove','tcp:%d'%port))
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 584, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '('adb', 'forward', '--remove', 'tcp:5628')' returned non-zero exit status 1`
```

dlenski, 2016-03-22:
Great! Ooh, so close.

Does `adb forward --list ` show this "leftover" TCP port forwarding which hasn't been removed properly?

I may have to add a delay or a loop to try to get rid of this error.

One of the reasons why I was hoping *not* to rely on TCP forwarding is that it seems to be very timing-dependent and prone to race conditions.

theduffmaster, 2016-03-23:
It does unfortunately.

dlenski, 2016-03-23:
Yes, it should be possible to remove the listener with the standard `adb forward --remove tcp:PORT`, which is what this program uses anyway.

It seems to be a timing issue; I don't have this problem on my Linux system so it's difficult for me to test, but I'll add a loop to retry removing the listener.

dlenski, 2016-03-23:
Closing this, listener issue to be continued in #3.

closed 2016-03-23 by dlenski