dlenski/tetherback #2: twrp_backup.py does not create or try to create data.ext4.win. It skips it completely.

theduffmaster, 2016-03-20:
Device: Nexus 5
OS: Mac OSX
It seems like it skips the data partition completely and doesn't even attempt to back it up and export it to a file. The nandroid_backup.py does not skip the data partition, but this program does. After backing up and saving the system partition the program ends with no reported errors.

`Device reports TWRP kernel (3.4.0-bricked-hammerhead-twrp-g7b77eb4).
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:02                                             
Saving TWRP backup images in twrp-backup-2016-03-20--18-35-43/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:03   2.95 MB/s                                 
Saving tarball of mmcblk0p25 (mounted at /system), 1024 MiB uncompressed...
  system.ext4.win: 100% Time: 0:03:06   2.94 MB/s  `

dlenski, 2016-03-20:
Thanks! Perhaps your `/data` partition is not named `userdata` in the block device list, for some reason?

I just pushed a new version with a couple command line options: `--verbose` to show the partition map and `--extra` to include an extra partition in the backup. Could you try these and let me know if they resolve this?

```
  partition map: 100% Time: 0:00:03                                                                                                                                             
	NAME	DEVICE	SIZE(MiB)	BACKUP?
1:	modem	mmcblk0p1	64		(skip)
2:	sbl1	mmcblk0p2	1		(skip)
3:	rpm	mmcblk0p3	0		(skip)
4:	tz	mmcblk0p4	0		(skip)
5:	sdi	mmcblk0p5	0		(skip)
6:	aboot	mmcblk0p6	0		(skip)
7:	pad	mmcblk0p7	2		(skip)
8:	sbl1b	mmcblk0p8	1		(skip)
9:	tzb	mmcblk0p9	0		(skip)
10:	rpmb	mmcblk0p10	0		(skip)
11:	abootb	mmcblk0p11	0		(skip)
12:	modemst1	mmcblk0p12	3		(skip)
13:	modemst2	mmcblk0p13	3		(skip)
14:	metadata	mmcblk0p14	0		(skip)
15:	misc	mmcblk0p15	16		(skip)
16:	persist	mmcblk0p16	16		(skip)
17:	imgdata	mmcblk0p17	3		(skip)
18:	laf	mmcblk0p18	22		(skip)
19:	boot	mmcblk0p19	22		boot
20:	recovery	mmcblk0p20	22		(skip)
21:	fsg	mmcblk0p21	3		(skip)
22:	fsc	mmcblk0p22	0		(skip)
23:	ssd	mmcblk0p23	0		(skip)
24:	DDR	mmcblk0p24	0		(skip)
25:	system	mmcblk0p25	1024		/system
26:	crypto	mmcblk0p26	30		(skip)
27:	cache	mmcblk0p27	700		(skip)
28:	userdata	mmcblk0p28	13089		/data
29:	grow	mmcblk0p29	0		(skip)
```

theduffmaster, 2016-03-21:
Definitely my fault for not being clear enough, but it is only the twrp_backup.py that gives my the partition skipped problem. Is there any chance you could add --verbose and --extra to the twrp_backup.py file?

dlenski, 2016-03-21:
Ah, this was just a dumb typo. I combined `twrp_backup.py` and `nandroid_backup.py` into one: `tetherback.py`.

Try this: `tetherback.py -n -v -t`

`-n`: make nandroid-style backup
`-v`: show partition table
`-t`: transfer files using TCP forwarding, rather than pipe

closed 2016-03-21 by dlenski

theduffmaster, 2016-03-22:
THANK YOU SO MUCH! It seems to be working now, and I'll try to see if the contents of the tarball come out right once I clear up some space on my computer. For the record I used -v and -t .
Here's the terminal output:
`Device reports TWRP kernel (3.4.0-bricked-hammerhead-twrp-g7b77eb4).
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:02                                             
--  --------  ----------  ----------  ------------------------------------
    NAME      DEVICE      SIZE (KiB)  BACKUP?
1   modem     mmcblk0p1   65536       (skip)
2   sbl1      mmcblk0p2   1024        (skip)
3   rpm       mmcblk0p3   512         (skip)
4   tz        mmcblk0p4   512         (skip)
5   sdi       mmcblk0p5   512         (skip)
6   aboot     mmcblk0p6   512         (skip)
7   pad       mmcblk0p7   2048        (skip)
8   sbl1b     mmcblk0p8   1024        (skip)
9   tzb       mmcblk0p9   512         (skip)
10  rpmb      mmcblk0p10  512         (skip)
11  abootb    mmcblk0p11  512         (skip)
12  modemst1  mmcblk0p12  3072        (skip)
13  modemst2  mmcblk0p13  3072        (skip)
14  metadata  mmcblk0p14  512         (skip)
15  misc      mmcblk0p15  16384       (skip)
16  persist   mmcblk0p16  16384       (skip)
17  imgdata   mmcblk0p17  3072        (skip)
18  laf       mmcblk0p18  22528       (skip)
19  boot      mmcblk0p19  22528       boot.emmc.win (raw image)
20  recovery  mmcblk0p20  22528       (skip)
21  fsg       mmcblk0p21  3072        (skip)
22  fsc       mmcblk0p22  512         (skip)
23  ssd       mmcblk0p23  512         (skip)
24  DDR       mmcblk0p24  512         (skip)
25  system    mmcblk0p25  1048576     system.ext4.win (tarball of /system)
26  crypto    mmcblk0p26  30720       (skip)
27  cache     mmcblk0p27  716800      (skip)
28  userdata  mmcblk0p28  28551146    data.ext4.win (tarball of /data)
29  grow      mmcblk0p29  5           (skip)
--  --------  ----------  ----------  ------------------------------------
Saving backup images in twrp-backup-2016-03-21--18-32-05/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.47 MB/s                                 
Saving tarball of mmcblk0p25 (mounted at /system), 1024 MiB uncompressed...
  system.ext4.win: 100% Time: 0:03:11   2.86 MB/s                               
Saving tarball of mmcblk0p28 (mounted at /data), 27881 MiB uncompressed...
  data.ext4.win: 100% Time: 3:33:57   1.73 MB/s               `

commit 5541ff84 (2016-03-21): fix #2
commit 9df5efbd (2016-06-25): fix #2