dlenski/tetherback #3: error: cannot remove listener

theduffmaster, 2016-03-22:
A small error that probably breaks nothing and is just python reminding you to keep your listeners tidy. Here it is in context with the full terminal output. It's the last couple lines:

```
Device reports TWRP kernel (3.4.0-bricked-hammerhead-twrp-g7b77eb4).
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:03                                             
--  --------  ----------  ----------  ---------------------------
    NAME      DEVICE      SIZE (KiB)  BACKUP?
1   modem     mmcblk0p1   65536       (skip)
2   sbl1      mmcblk0p2   1024        (skip)
3   rpm       mmcblk0p3   512         (skip)
4   tz        mmcblk0p4   512         (skip)
5   sdi       mmcblk0p5   512         (skip)
6   aboot     mmcblk0p6   512         (skip)
7   pad       mmcblk0p7   2048        (skip)
8   sbl1b     mmcblk0p8   1024        (skip)
9   tzb       mmcblk0p9   512         (skip)
10  rpmb      mmcblk0p10  512         (skip)
11  abootb    mmcblk0p11  512         (skip)
12  modemst1  mmcblk0p12  3072        (skip)
13  modemst2  mmcblk0p13  3072        (skip)
14  metadata  mmcblk0p14  512         (skip)
15  misc      mmcblk0p15  16384       (skip)
16  persist   mmcblk0p16  16384       (skip)
17  imgdata   mmcblk0p17  3072        (skip)
18  laf       mmcblk0p18  22528       (skip)
19  boot      mmcblk0p19  22528       boot.img.gz (raw image)
20  recovery  mmcblk0p20  22528       (skip)
21  fsg       mmcblk0p21  3072        (skip)
22  fsc       mmcblk0p22  512         (skip)
23  ssd       mmcblk0p23  512         (skip)
24  DDR       mmcblk0p24  512         (skip)
25  system    mmcblk0p25  1048576     system.img.gz (raw image)
26  crypto    mmcblk0p26  30720       (skip)
27  cache     mmcblk0p27  716800      (skip)
28  userdata  mmcblk0p28  28551146    userdata.img.gz (raw image)
29  grow      mmcblk0p29  5           (skip)
--  --------  ----------  ----------  ---------------------------
Saving backup images in nandroid-backup-2016-03-21--18-48-09/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.img.gz: 100% Time: 0:00:11 844.82 kB/s                                   
Saving partition system (mmcblk0p25), 1024 MiB uncompressed...
  system.img.gz: 100% Time: 0:10:17 944.49 kB/s                                 
Saving partition userdata (mmcblk0p28), 27881 MiB uncompressed...
  userdata.img.gz: 100% Time: 3:48:44   1.69 MB/s                                                                                                                                                                                             
error: cannot remove listener
error: cannot remove listener
Traceback (most recent call last):
  File "/Volumes/GLYPH-2 TB/nexus5recovery/tetherback.py", line 125, in <module>
    sp.check_call(('adb','forward','--remove','tcp:%d'%port))
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 584, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '('adb', 'forward', '--remove', 'tcp:5628')' returned non-zero exit status 1

```

dlenski, 2016-03-23:
#

theduffmaster, 2016-03-25:
I think it's fixed now, I don't get the error anymore. But you never know with these listener errors, they always seem to come back even when you think you've fixed them.

dlenski, 2016-03-25:
Good to know, thanks! This is why I was hesitant to use the TCP forwarding in the first place… I will close it for now but will try to come up with some more robust workarounds as well.

commit 670c2275 (2016-03-23): retry removal of adb forwarding -- does this fix #3?
commit f56f6385 (2016-06-25): retry removal of adb forwarding -- does this fix #3?
closed 2016-03-25 by dlenski