dlenski/tetherback #5: missing tetherrestore

kerlerm, 2016-04-08:
Hi Dan. 

Cool backup tool. I'm missing a "tetherrestore". :-)
How do I get the contents back onto the phone when I can not copy them over and flash them due to storage space constraints (=same problem inverted)?

Perhaps you could expand on this in the README section? Thanks!

Martin

dlenski, 2016-04-08:
Excellent question. I was wondering when this would come up :-)

I haven't pushed any restore code out yet because it's a lot more time-consuming to verify that it's working correctly in all cases. You are right that it is basically the reverse of the backup case (untar or dd).

Let me see if I can put something together this weekend.

paambaati, 2016-06-11:
I'd love this too. In the meanwhile, how is everyone else using these backups?

dlenski, 2016-06-11:
I have done a few "manual" restores by piping the backup over `adb` and executing `tar` on the phone side. (Basically, just what tetherback does in reverse.)

I do hope to write a full-fledged restore tool eventually.

In the meantime, if you have code to contribute, please submit pull requests :+1:

pati610, 2016-06-25:
Looks very promising! I get an error, though, "OverflowError: normalized days too large to fit in a C int", issued by the function format_time in utils.py, line 222. Any ideas why?

Patrick
----------------------

Device reports TWRP version 2.8.6.1
Reading partition map for mmcblk0 (47 partitions)...
  partition map: 100% Time: 0:00:07                                            
Saving backup images in ./twrp-backup-2016-06-25--06-10-28/ ...
Saving partition boot (mmcblk0p42), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.4 MiB/s   7.5 MiB                      
Saving tarball of mmcblk0p45 (mounted at /system), 2688 MiB uncompressed...
Traceback (most recent call last):
  File "tetherback.py", line 255, in <module>
    pbar = ProgressBar(max_value=size*512, widgets=pbwidgets).start()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 539, in start
    self.update(self.min_value, force=True)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 496, in update
    StdRedirectMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 131, in update
    DefaultFdMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 42, in update
    self.fd.write('\r' + self._format_line())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 433, in _format_line
    widgets = ''.join(self._format_widgets())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 413, in _format_widgets
    widget_output = widget(self, data)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/widgets.py", line 289, in __call__
    data['eta'] = utils.format_time(data['eta_seconds'])
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/utils.py", line 222, in format_time
    time = datetime.timedelta(seconds=int(time))
OverflowError: normalized days too large to fit in a C int
Patis-iMac:tetherback pati$ python3 tetherback.py
Device reports TWRP version 2.8.6.1
Reading partition map for mmcblk0 (47 partitions)...
  partition map: 100% Time: 0:00:07                                            
Saving backup images in ./twrp-backup-2016-06-25--06-20-32/ ...
Saving partition boot (mmcblk0p42), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:01   4.1 MiB/s   7.5 MiB                      
Saving tarball of mmcblk0p45 (mounted at /system), 2688 MiB uncompressed...
Traceback (most recent call last):
  File "tetherback.py", line 255, in <module>
    pbar = ProgressBar(max_value=size*512, widgets=pbwidgets).start()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 539, in start
    self.update(self.min_value, force=True)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 496, in update
    StdRedirectMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 131, in update
    DefaultFdMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 42, in update
    self.fd.write('\r' + self._format_line())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 433, in _format_line
    widgets = ''.join(self._format_widgets())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 413, in _format_widgets
    widget_output = widget(self, data)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/widgets.py", line 289, in __call__
    data['eta'] = utils.format_time(data['eta_seconds'])
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/utils.py", line 222, in format_time
    time = datetime.timedelta(seconds=int(time))
OverflowError: normalized days too large to fit in a C int
-------------------------

dlenski, 2016-08-21:
I've started working on this in the `tetherrestore` branch: https://github.com/dlenski/tetherback/tree/feature/tetherrestore

@kerlerm, @pati610: Could you please try `test_restore_partition.py` and let me know if this completes without error? It's just a quick and dirty test to generate a file full of random data and then push it to the device using the new method of `restore_partition()`.

There are some timing issues that I ran into, and also it appears that `tetherrestore` is only going to be practical with newer ADB and TWRP which support `adb exec-in`.`adb shell` simply doesn't work as an input pipe.

kerlerm, 2016-08-21:
Here is the output - it completed without error on twrp 3.0.2 (maguro)
```
$ tests/test_restore_partition.py /tmp/tmp1pesmyye 512178
ADB: ('adb', 'shell', 'rm -f /tmp/md5in /tmp/md5out 2> /dev/null; mknod /tmp/md5in p')
Restoring image of partition test (../../tmp/test.bin), 0 MiB uncompressed...
ADB: ('adb', 'shell', 'umount /dev/block/../../tmp/test.bin 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount None 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount /dev/block/../../tmp/test.bin 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount None 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount /dev/block/../../tmp/test.bin 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount None 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount /dev/block/../../tmp/test.bin 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'umount None 2>/dev/null && echo ok')
ADB: ('adb', 'shell', 'mount')
ADB: ('adb', 'shell', 'md5sum /tmp/md5in > /tmp/md5out')
ADB: ('adb', 'exec-in', 'tee /tmp/md5in | gunzip -f | dd of=/dev/block/../../tmp/test.bin 2> /dev/null')
  /tmp/tmp1pesmyye: 100%  49.2 MiB/s 500.2 KiB                                                                                            
ADB: ('adb', 'shell', 'cat /tmp/md5out && rm -f /tmp/md5in /tmp/md5out')
```

dlenski, 2016-08-21:
Excellent, thanks @kerlerm!

paulie-g, 2016-09-10:
Presumably restoring tetherback's twrp-style backup with twrp itself works?

paambaati, 2016-09-10:
> Presumably restoring tetherback's twrp-style backup with twrp itself works?

@paulie-g You mean mount the backup directory on TWRP via MTP and use it? If yes, that is great news!

paulie-g, 2016-09-10:
No, I mean exactly what I said - taking the twrp-style backup tetherback creates, putting it on internal storage say and restoring it with twrp. It was also a question rather than a statement ;)

jamy015, 2016-10-29:
Has anyone verified whether @paulie-g's theory works?

If not, what's 'TWRP-style' about the TWRP-style backups? Would it even theoretically be possible for this to work? (@dlenski)

I can try to test this myself, but I don't have any phones except my main one. If somebody else has a spare phone to test with, I'd like to not potentially wreck my only phone 😛

Blender3D, 2016-10-31:
@jamy015 I've done it a few times on my unencrypted Nexus 5 and it's works fine.

dlenski, 2016-11-01:
@Blender3D  @jamy015,

I have restored from both the TWRP-style and Nandroid-style backups as well, also on an unencrypted Nexus 5 as well as once on a Galaxy S4. (I would feel pretty guilty about calling this a *backup* tool if it were **impossible** to use the backups to restore :-p)

There are a couple times when restoring the `/system` partition from the TWRP/tarball backup **did not work** and resulted in a boot loop. However, restoring the `/data` partition from a TWRP/tarball backup has worked perfectly, as has restoring the `/system` partition from a Nandroid backup.

Blender3D, 2016-11-01:
@dlenski I made a backup of my `/system` partition with both TWRP and tetherback and the only differences between the two tar archives are:

 - tetherback includes `lost+found` and the root folder.
 - tetherback uses relative paths.
 - TWRP uses something like [`--absolute-names`](https://www.gnu.org/software/tar/manual/html_node/absolute.html) and [ignores a few files](https://github.com/TeamWin/Team-Win-Recovery-Project/blob/f177066df714a9a8b40daca80e922509da5e8af4/twrpDU.cpp#L99-L115).

Could the relative/absolute paths break something?

dlenski, 2016-11-03:
@Blender3D, good observations. I believe that including `lost+found` doesn't matter, nor that relative vs. absolute paths.

I thought the problem might have something to do with not correctly saving and restoring the [SELinux context](https://linux.die.net/man/1/chcon); the `tar -p` option with Busybox `tar` is supposed to make it save and restore the SELinux context, but documentation on this option is sparse and I'm honestly unsure if I'm using it correctly.

misterpyrrhuloxia, 2018-02-11:
So can someone tell me what command to run to restore my /data parition?

timikesmarky, 2019-02-10:
@dlenski Hello, my nexus 5's wifi chip stopped working suddenly. I want to restore persist, misc and EFS partitions. Can you please tell me how to do it? I have old backups in .img format taken with partition backup and restore app. This app says it restored but it's not successful. I think the partition needs to be unmounted first to restore. I've been searching since morning how to restore. I hope you can help me. Thanks!

edrex, 2019-09-26:
I just stumbled on tetherback after having used exec-out to backup /data/media a few times (as detailed  [here](https://tadej.ja.nez.si/android-internal-storage-backup.html)).

I'm finding that with my adb (android-tools 9.0.0_r45) and latest TWRP exec-in doesn't seem to work. Hopefully just something with my config and not irreversible bitrot on these undocumented commands :/

vonpupp, 2020-05-13:
Hello,

Tetherback is an awesome idea. However I stumbled with this issue and I feel I am a bit trapped. I backed up everything on my phone, about 55Gb but my phone is 64Gb so I cannot copy the backup on the phone and restore it directly on TWRP since there is not enough space.

Unless I cannot do a restore without the backup touching the phone, I don't think I will be able to restore anything at this point; which is a bit frustrating to be honest.

Could you please point me to any hack that would allow me to restore TWRP backup style without copying the backup to the phone? I don't have a pendrive where I am at right now but In about half an hour I will try with one.

Thanks.

vonpupp, 2020-05-13:
Ok, now I am in fact worried...

I have taken a backup with `tetherback -t -M -D -R -C`. I copied it to a USB SSD drive formated as NTFS and booted into TWRP.

I could mount the drive and select the folder and I clicked restore, but now my phone seems frozen, I cannot even get to boot as recovery (TWRP).

Any idea on how to fix this please?

Thanks

ulph0, 2022-09-02:
hi, i am trying to backup a phone with broken screen (galaxy 5) to a new one of the same model. 

i tried it with 

adb shell twrp backup SDBOM

but that only gives me a

TWRP does not appear to be running. Waiting for TWRP to start . . .
Press CTRL + C to quit

thetherback without otions gives me a error after generating boot.emmc.win:

```
Traceback (most recent call last):
  File "/home/ulf/.local/bin/tetherback", line 8, in <module>
    sys.exit(main())
  File "/home/ulf/.local/lib/python3.10/site-packages/tetherback/tetherback.py", line 359, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/home/ulf/.local/lib/python3.10/site-packages/tetherback/tetherback.py", line 245, in backup_partition
    raise RuntimeError("%s: don't know how to mount this partition" % pi.devname)
RuntimeError: mmcblk0p23: don't know how to mount this partition
```
while the nandroid version seems to backup eveything without problems: but as i am not so into details of adb, twrp and phoneOS i don't have a clue how to restore that. please give me a hint.