dlenski/tetherback #9: add corruption checks

kerlerm, 2016-04-17:
Each backup file should get its md5sum stored as TWRP does. Perhaps set up as an option. This makes it easier to spot corrupted images.

I suspect corruption in my backup because the backup of /data is not the size it should be. Another check could compare the amount of bytes backed up with the amount of used bytes on the mounted partition. 

Would that be helpful? 
Why does TWRP split big backup files (.000, .001...) anyway? Should the script behave the same?

kerlerm, 2016-04-17:
Hi Dan. Perhaps this (undocumented?) feature of ADB will solve this CRLF corruption issue?

https://plus.google.com/110558071969009568835/posts/Ar3FdhknHo3

dlenski, 2016-04-18:
Very nice! That would indeed solve a lot of problems.

As for md5sums, this has been on my to-do list. The reason I haven't added them already is because the md5sum should ideally be computed on *both* the device and the host, then compared to verify no corruption in transmission. I haven't figured out a simple and efficient way to do this yet.

dlenski, 2016-04-18:
@kerlerm, does `adb exec-out` actually work for you? I always get `error: closed` when trying to execute commands with it (adb 1.0.32).

kerlerm, 2016-04-18:
Which parameters do you pass to adb?

dlenski, 2016-04-19:
I just tried a few simple examples like `adb exec-out ls -R /`. They all
resulted in the same error message for me, even though `adb shell` works
fine with the same executable.

I saw several others had reported the same issue on koush's G+ post.

kerlerm, 2016-04-19:
Hi Dan. 

I think it's more a pipe-like thing. The C code shows something called "fd". Is this a file descriptor? 

So this doesn't execute but opens a binary link from android to the host.

Perhaps we should write to the guy who changed adb code in 2015 to add exec-in/out and ask for clarification.

Martin

kerlerm, 2016-04-19:
I'm running latest adb on linux, just updated from devel.google to make sure that the exec-in/out patch is included:
Android Debug Bridge version 1.0.32
Revision 09a0d98bebce-android

$ adb exec-out dd if=/storage/emulated/legacy/scrshot.png > scrshot.png
...transfers the image without corruption to my host.

Does that work in windows?

dlenski, 2016-04-19:
(I don't actually have a computer running Windows handy ☺)

This is quite strange: `exec-out` *does* work for me from the TWRP recovery `adbd`, but it does *not* work with the Cyanogenmod 10 (Android 4.4.1/KitKat) version of `adbd`. Very odd.

I'll add it as an extra option. It's extremely frustrating that there is no option which seems to work across many versions of adb on both the host and device :-1:

kerlerm, 2016-04-19:
I don't have windows here either. :)

Concerning the different behaviour of adb: Are you sure that this exec-out patch of summer 2015 made it into CM10 Android 4.4? Perhaps it is still missing there.

Concerning TCP timeout issues: I read a recommendation to leave the phone in active mode during transfer because otherwise the reduced processor cycle in suspend could lead to lost packages. Can't believe this could be true!?

dlenski, 2016-04-19:
> Are you sure that this exec-out patch of summer 2015 made it into CM10 Android 4.4? Perhaps it is still missing there.

This is probably the right answer.

It is not totally clear to me whether `exec-out` requires a change *only* on the host side or also on the device side.

kerlerm, 2016-04-24:
You got me wrong: adb exec-* is *not* working with TWRP 2.8.0 for me.
I tested with android OS running, so I was checking the ADB included in the recent CM12.1 build (client side) and adb from latest google git (host side). This did work.

Server and client seem to negotiate a common feature set. So exec-in/exec-out may not be present on both sides, especially in older TWRP builds:

Adb server version (1.0.ADB_SERVER_VERSION) got bumped several times:

2012-08-31 21:31:36: bump to 31: commit 64b3103017cb9038c5fb7e3601f51c6a458bed06
--- 2014-05-30 11:27:56: exec-out/-in patch added ---
2014-07-07 16:51:58: bump to 32: commit 71fe584a1a7256c057267ae46a35f7c74d1d549e

So all adb builds >= 1.0.32 should have a working exec-out.

Do you know how to extract adb version from TWRP?

dlenski, 2016-04-24:
Ah, thanks for clarifying this. For me, exec-out *did* work with TWRP
v3.0.0.

Have you tried that version with your N4?

Also, does the TWRP style backup actually work while booted into the full
Android OS? I'm a bit leery about the possibility of corruption due to
files changing on the `/data` partition while the backup is running.

I do not know how to get the adb version from TWRP. I believe it is based
on the omnirom-recovery codebase but I don't totally understand the
relationship between the two projects.

kerlerm, 2016-04-24:
The N4 is my wife's phone - so only limited testing time available :-)
She's running TWRP 2.8.5.2 at the moment - I'll test that later...

I did not run tetherback.py with a fully booted Android OS. Just commandline tests. Data corruption would make the backup unusable. I second that.

kerlerm, 2016-04-24:
Nexus 4: exec-out-backup seems to work fine with TWRP 2.8.5.2 and adb 1.0.32. I disabled the faulty adb version check, though.

`./tetherback.py -x -v  -R -M -o /home/nas/backup/android/mako/` 
I have not compared md5sums yet!

So when using the -x option, should we check for adb >=1.0.32 on the host or wait for more feedback?

dlenski, 2016-04-25:
> So when using the -x option, should we check for adb >=1.0.32 on the host or wait for more feedback?

I wish I had a good answer here. Checking for >=1.0.32 seems prudent.

If there was a single transfer method that was both fast and universally available across adb/recovery versions, I'd just turn off everything else. >:-(

dlenski, 2016-05-08:
@kerlerm, please take a look at 8242710f336e75b08eb2e9a7776dabcc35705988: I added md5sum generation of the device side and verification on the host side, which should be a very strong guarantee against corruption.

Does this work as expected for you?

closed 2016-05-08 by dlenski

kerlerm, 2016-05-09:
Thanks! I will do so when the Nexus 4 is working again (broken front panel) *sigh*.

kerlerm, 2016-05-24:
The Nexus4 still broken. I tested with TWRP 3.0.2 on picassowifi:

```
boot.emmc.win: 100% Time: 0:00:00   5.5 GiB/s                                                                                                                                    
Traceback (most recent call last):
  File "./tetherback.py", line 261, in <module>
    devicemd5 = sp.check_output(adbcmd+('shell','cat /tmp/md5out')).strip().decode().split()[0]
IndexError: list index out of range
```

dlenski, 2016-05-24:
@kerlerm, interesting… that means that `/tmp/md5out` is empty.

Does the picassowifi version of TWRP have `busybox md5sum` included?

* If so, I can't understand how the `/tmp/md5out` file would end up empty. You'd have to try stepping through the individual device-side commands one-by-one.
* If not— and this is probably inevitable—I'm going to have to start adding checks for the existence of every single `busybox` tool used by the script.

Also, the addition of the `*` before the filename **is** intentional. The GNU coreutils version of `md5sum` adds a `*` before the filename when run as `md5sum --binary`. This is so that the file will not be misinterpreted as a text file on subsequent verification, possibly causing a mismatched hash due to CRLF translation. (Fun stuff :8ball:)

z2z, 2016-11-27:
Update, On windows 7 with Motorola moto turbo (xt1255 ~ quark) 

```
D:\turbo>tetherback
tetherback-script.py v0.9.1
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.85-perf-gf50d607
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (41 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%
Saving backup images in .\twrp-backup-2016-11-27--16-34-39/ ...
Saving partition boot (mmcblk0p36), 16 MiB uncompressed...
  boot.emmc.win: 100%   3.6 MiB/s   8.7 MiB
Traceback (most recent call last):
  File "C:\Python34\Scripts\tetherback-script.py", line 9, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "c:\python34\lib\site-packages\tetherback\tetherback.py", line 344, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "c:\python34\lib\site-packages\tetherback\tetherback.py", line 293, in backup_partition
    raise RuntimeError("md5sum mismatch (local %s, device %s)" % (localmd5, devicemd5))
RuntimeError: md5sum mismatch (local b804df2473afbe6d58bf44e8832e668b, device 84c3f4252beaa6abc93e7f12ef65291a)
```

`tetherback -t `  in progress, no errors so far ..
```
D:\turbo>tetherback -t
tetherback-script.py v0.9.1
Found ADB version 1.0.32
Device reports kernel 3.10.85-perf-gf50d607
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (41 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%
Saving backup images in .\twrp-backup-2016-11-27--16-36-42/ ...
Saving partition boot (mmcblk0p36), 16 MiB uncompressed...
  boot.emmc.win: 100%   4.2 MiB/s   8.7 MiB
Saving tarball of mmcblk0p40 (mounted at /system), 3376 MiB uncompressed...
  system.ext4.win: 100%   4.6 MiB/s   1.3 GiB
Saving tarball of mmcblk0p41 (mounted at /data), 54543 MiB uncompressed...
  data.ext4.win:  20%   4.5 MiB/s  10.8 GiB
```

dlenski, 2016-11-28:
> Update, On windows 7 with Motorola moto turbo (xt1255 ~ quark)
>
> D:\turbo>tetherback
> tetherback-script.py v0.9.1
> Found ADB version 1.0.32
> Using default transfer method: adb exec-out pipe (--exec-out) [...]
'tetherback')()
>   File "c:\python34\lib\site-packages\tetherback\tetherback.py", line
344, in main
>     backup_partition(adb, partmap[standard], bp, args.transport,
backupdir, args.verify)
>   File "c:\python34\lib\site-packages\tetherback\tetherback.py", line
293, in backup_partition
>     raise RuntimeError("md5sum mismatch (local %s, device %s)" %
(localmd5, devicemd5))
> RuntimeError: md5sum mismatch (local b804df2473afbe6d58bf44e8832e668b,
device 84c3f4252beaa6abc93e7f12ef65291a)
>
> tetherback -t in progress, no errors so far ..

This is a frustrating one. I don't think I've seen another case where `adb
exec-out` with v1.0.32 results in a corrupt transfer, but TCP doesn't.

Can you manually run `adb exec-out "cat /dev/block/mmcblk0p36" > temp_file
and do a binary diff against the version produced with TCP transfer, in
order to figure out where the corruption is occurring?

dlenski, 2016-11-28:
> Update, On windows 7 with Motorola moto turbo (xt1255 ~ quark)
>
> D:\turbo>tetherback
> tetherback-script.py v0.9.1
> Found ADB version 1.0.32
> Using default transfer method: adb exec-out pipe (--exec-out) [...]

Hi @z2z, this is a frustrating one. I don't think I've seen another case where `adb
exec-out` with v1.0.32 results in a corrupt transfer, but TCP doesn't.

Can you manually run `adb exec-out "cat /dev/block/mmcblk0p36" > temp_file
and do a binary diff against the version produced with TCP transfer, in
order to figure out where the corruption is occurring?

commit 8242710f (2016-05-08): record md5sums on Android device side and verify them on host side, for a strong corruption check (fixes #9)