dlenski/tetherback #12: progress bar fails

kerlerm, 2016-04-28:
Progressbar2 fails for me. I'm sure it worked before...

```
Saving partition boot (mmcblk0p7), 8 MiB uncompressed...                                                                                                                            
Traceback (most recent call last):                                                                                                                                                  
  File "./tetherback.py", line 243, in <module>                                                                                                                                     
    pbar.maxval = out.tell() or pbar.maxval # need to adjust for the smaller compressed size
AttributeError: 'ProgressBar' object has no attribute 'maxval'
```

dlenski, 2016-04-28:
Hrrrmmm... it looks like `maxval` has been changed to `max_value` in recent versions of progressbar2 and is now deprecated as an alias.
https://github.com/WoLpH/python-progressbar/blob/0d35e8b6b1f7501198f0ca19309df5a0260201a3/progressbar/bar.py#L179-L182

What version of progressbar2 are you using? I'm using the (apparently ancient) v2.3 [as provided by Ubuntu](http://packages.ubuntu.com/wily/python/python3-progressbar).

I've been meaning to put together a proper `setup.py` so that tetherback can be installed with `pip` and pull in the right dependencies and versions.

kerlerm, 2016-04-29:
Hi Dan.

Mine is progressbar2 (3.6.2) according to `pip list`

jrussell88, 2016-05-18:
I have the same issue with Ubuntu 15.10 and Python 3.4.3-4ubuntu1. 

Versions are:
- progressbar (2.3)
- progressbar2 (3.6.2)

I changed the two instances of `maxval=` to `max_value=`
and `pbar.maxval` to `pbar.max_value`

dlenski, 2016-06-07:
This should be fixed in eed471c0174cd85b60557ccd299793b447a2eb5c, which has a proper `setup.py` and specifically requires `progressbar2`.

closed 2016-06-07 by dlenski

sehargul-123, 2019-11-04:
(venv) sehar@sehar-HP-Z220-CMT-Workstation:~/DeepSpeech$ python -u DeepSpeech.py --noshow_progressbar   --train_files urdu-train.csv   --test_files urdu-train.csv   --train_batch_size 1   --test_batch_size 1   --n_hidden 100   --epochs 200   --checkpoint_dir checkpoint  --alphabet_config_path=data/lm/alphabet.txt
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/sehar/DeepSpeech/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/sehar/venv/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
WARNING:tensorflow:From /home/sehar/DeepSpeech/tensorflow/python/data/ops/dataset_ops.py:494: py_func (from tensorflow.python.ops.script_ops) is deprecated and will be removed in a future version.
Instructions for updating:
tf.py_func is deprecated in TF V2. Instead, there are two
    options available in V2.
    - tf.py_function takes a python function which manipulates tf eager
    tensors instead of numpy arrays. It's easy to convert a tf eager tensor to
    an ndarray (just call tensor.numpy()) but having access to eager tensors
    means `tf.py_function`s can use accelerators such as GPUs as well as
    being differentiable using a gradient tape.
    - tf.numpy_function maintains the semantics of the deprecated tf.py_func
    (it is not differentiable, and manipulates numpy arrays). It drops the
    stateful argument making all functions stateful.
    
W1104 13:42:51.006331 139628039710528 deprecation.py:323] From /home/sehar/DeepSpeech/tensorflow/python/data/ops/dataset_ops.py:494: py_func (from tensorflow.python.ops.script_ops) is deprecated and will be removed in a future version.
Instructions for updating:
tf.py_func is deprecated in TF V2. Instead, there are two
    options available in V2.
    - tf.py_function takes a python function which manipulates tf eager
    tensors instead of numpy arrays. It's easy to convert a tf eager tensor to
    an ndarray (just call tensor.numpy()) but having access to eager tensors
    means `tf.py_function`s can use accelerators such as GPUs as well as
    being differentiable using a gradient tape.
    - tf.numpy_function maintains the semantics of the deprecated tf.py_func
[... cut ...]
  File "DeepSpeech.py", line 890, in main
    train()
  File "DeepSpeech.py", line 608, in train
    train_loss, _ = run_set('train', epoch, train_init_op)
  File "DeepSpeech.py", line 566, in run_set
    pbar = create_progressbar(prefix=prefix, widgets=widgets, suffix=suffix).start()
  File "/home/sehar/DeepSpeech/util/logging.py", line 44, in create_progressbar
    return progressbar.NullBar(*args, **kwargs)
AttributeError: module 'progressbar' has no attribute 'NullBar'
kindly help in resolving this issue

sehargul-123, 2019-11-04:
Mine progressbar=2.5
progressbar2=3.47.0
ubuntu 18.04