dlenski/tetherback #13: twrp 2.8.1.0 reports different name than -twrp-

kerlerm, 2016-04-28:
My TWRP version 2.8.1.0 reports -cyanogenmod- instead of -twrp-. Kernel check code should be adapted.

dlenski, 2016-04-28:
Argh... it seems that the only real way to solve these issues is going to be maintaining a whitelist of which exact recovery versions and host `adb` versions work with which transfer methods.

Is there any way for `adb` to tell me that the device it is connected to is definitely in recovery mode rather than Android system mode?

Really, the only version to check the kernel is to try to confirm that the device is in recovery mode—because trying to backup a live RW-mounted `/data` partition is likely to result in data corruption as we've discussed previously.

kerlerm, 2016-04-29:
Doesn't `adb devices` show the state the device is in?
`$ adb devices`
`<ID> recovery`

jrussell88, 2016-05-07:
Likewise TWRP 2.8.7.0 reports CM

> tetherback.py:

 error: device reports non-TWRP kernel (3.4.67-cyanogenmod-gaff3758)

jrussell88, 2016-05-07:
I see you're using the command 
> uname -r
to check the kernel version.

The command:
> twrp -v
returns the version of TWRP, which may solve the problem.

dlenski, 2016-05-07:
Aha! Thank you very much, @jrussell88. This is a much better way to check for TWRP recovery and the specific version.

commit dbb6a4d2 (2016-05-08): use twrp -v to check TWRP version (fix #13)
closed 2016-05-08 by dlenski