dlenski/tetherback #14: upper/lowercase for partition names

kerlerm, 2016-05-09:
Device "picassowifi" with TWRP 2.8.7.0 reports all partitions in upper case. So testing lowercase only does not work.

dlenski, 2016-05-09:
Huh... Interesting. What version of Android is this running? Is it a stock device or running a custom ROM?

https://wiki.cyanogenmod.org/w/Picassowifi_Info

dlenski, 2016-05-09:
Can you send me the output of `tetherback -0` showing the partition map for this device?

kerlerm, 2016-05-09:
Hi Dan. This is cm-12.1-20160501-nightly with twrp 2.8.7.0

kerlerm, 2016-05-09:
Device reports TWRP version 2.8.7.0
Reading partition map for mmcblk0 (21 partitions)...
  partition map: 100% Time: 0:00:01

BLOCK DEVICE    NAME          SIZE (KiB)    FILENAME    FORMAT
--------------  ----------  ------------  ----------  --------
mmcblk0p1       BOTA0               4096
mmcblk0p2       BOTA1               4096
mmcblk0p3       EFS                20480
mmcblk0p4       m9kefs1             4096
mmcblk0p5       m9kefs2             4096
mmcblk0p6       m9kefs3             4096
mmcblk0p7       CARRIER             4096
mmcblk0p8       PARAM               8192
mmcblk0p9       BOOT                8192
mmcblk0p10      RECOVERY           10240
mmcblk0p11      OTA                10240
mmcblk0p12      CDMA-RADIO          4096
mmcblk0p13      RADIO              20480
mmcblk0p14      TOMBSTONES          4096
mmcblk0p15      TDATA               4096
mmcblk0p16      PERSDATA           12288
mmcblk0p17      RESERVED2           4096
mmcblk0p18      SYSTEM           2560000
mmcblk0p19      CACHE             204800
mmcblk0p20      HIDDEN            256000
mmcblk0p21      USERDATA        12226560

dlenski, 2016-05-09:
That is strange. Okay, case-insensitive matching it is...

commit 0bc466eb (2016-05-09): fix #14 by lowercase-ing partition names from /sys
closed 2016-05-09 by dlenski