dlenski/tetherback #16: Files lost in compression

jrussell88, 2016-05-18:
With Ubuntu 15.10 x64 and Python3 3.4.3-4ubuntu1 the saved files are empty or incomplete although the programme appears to run correctly. Repeating the backup produces the same results:
`john@johns-MacBookPro:/Google Nexus 4 - mako/Backups/TWRP Nandroid backups/004b1e708d8d9655$ /media/john/Home/john/Documents/Computers/Android/Tetherback/tetherback.py
Device reports TWRP kernel (/system/bin/sh: twrp: not found).
Reading partition map for mmcblk0 (25 partitions)...
  partition map: 100% Time: 0:00:03                                                                                                                                                                                                            
Saving backup images in ./twrp-backup-2016-05-18--14-55-30/ ...
Saving partition boot (mmcblk0p6), 22 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:00   7.6 GiB/s                                                                                                                                                                                                
Saving tarball of mmcblk0p21 (mounted at /system), 840 MiB uncompressed...
  system.ext4.win: 100% Time: 0:00:00 402.5 GiB/s                                                                                                                                                                                              
Saving tarball of mmcblk0p23 (mounted at /data), 13475 MiB uncompressed...
  data.ext4.win: 100% Time: 0:00:00   6.2 TiB/s `
File sizes for all three partitions are zero.

and using the -P flag, 452Mb for the system partition, which extracts to 733Mb; 20 to 170 bytes for the other three which appear to be empty:
`john@johns-MacBookPro:/Google Nexus 4 - mako/Backups/TWRP Nandroid backups/004b1e708d8d9655$ /media/john/Home/john/Documents/Computers/Android/Tetherback/tetherback.py -MRP
WARNING: binary pipe (--pipe) may CORRUPT DATA on non-Linux host
Device reports TWRP kernel (/system/bin/sh: twrp: not found).
Reading partition map for mmcblk0 (25 partitions)...
  partition map: 100% Time: 0:00:03                                                                                                                                                                                                            
Saving backup images in ./twrp-backup-2016-05-18--14-55-59/ ...
Saving partition boot (mmcblk0p6), 22 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:00 303.7 B/s                                                                                                                                                                                                  
Saving partition recovery (mmcblk0p7), 22 MiB uncompressed...
  recovery.emmc.win: 100% Time: 0:00:00 298.4 B/s                                                                                                                                                                                              
Saving tarball of mmcblk0p21 (mounted at /system), 840 MiB uncompressed...
  system.ext4.win: 100% Time: 0:07:02   1.0 MiB/s                                                                                                                                                                                              
Saving tarball of mmcblk0p23 (mounted at /data), 13475 MiB uncompressed...
  data.ext4.win: 100% Time: 0:00:00   2.6 KiB/s                                                                                                                                                                                                
john@johns-MacBookPro:/Google Nexus 4 - mako/Backups/TWRP Nandroid backups/004b1e708d8d9655$ `

dlenski, 2016-05-18:
What recovery are you running?

`/system/bin/sh: twrp: not found`

dlenski, 2016-05-18:
It appears that you modified the program to skip the TWRP recovery version check. Unless I'm missing something...?

Can you run again with `-v`?

TWRP is not optional. It is the only recovery, that I know of, that has the right version of `tar` to be able to back up `/data` and `/system` including SELinux permissions.

jrussell88, 2016-05-20:
Thanks for the answer - I'm running TWRP 3.0.2. As I explained elsewhere a kernel test for TWRP  is unreliable, so I edited the code to run when it fails the test rather than when it passes.

dlenski, 2016-05-20:
Yes, but I'm not sure which version of `tetherback` you've modified.

The latest version does *not* do a TWRP kernel check; it runs `twrp --version` ([as you suggested](https://github.com/dlenski/tetherback/issues/13#issuecomment-217656245) on issue #13).

The latest version also checks the md5sum on both host and device side to verify correct data transfer (8242710f336e75b08eb2e9a7776dabcc35705988).

The output that you showed indicates a *kernel* version check. So I do not think it is based on my latest version of `tetherback`, and may not include the data transfer check.

Can you please fork and push your modified version so we can compare them? Without knowing exactly what's different about your code, it's hard to understand where the problem might be.

dlenski, 2016-06-08:
@jrussell88, are you still having this issue? I haven't been able to reproduce it in any way, and don't know exactly what version you're running.

closed 2016-06-10 by dlenski