dlenski/tetherback #17: WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''

Joyomojo, 2016-06-03:
http://pastebin.com/dYiADhFa

Device Xiaomi Mi4
WIn10
Python 3.5.1
TWRP 3.0.2-0
ADB 1.0.32

Script only creates an empty backup folder.

Arno500, 2016-06-05:
Just a "me too!" on OnePlus One.
Here is the log:
`$ python3 tetherback.py -vMDRCx --force
Found ADB version 1.0.35
Device reports TWRP version 3.0.2-0
WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/uevent': ''
Reading partition map for mmcblk0 (29 partitions)...
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p1/uevent': ''
  partition map:   3% ETA:  0:00:05                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p2/uevent': ''
  partition map:   6% ETA:  0:00:04                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p3/uevent': ''
  partition map:  10% ETA:  0:00:04                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p4/uevent': ''
  partition map:  13% ETA:  0:00:04                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p5/uevent': ''
  partition map:  17% ETA:  0:00:03                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p6/uevent': ''
  partition map:  20% ETA:  0:00:03                                                                                                                                                                              WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
WARNING: don't understand this line from '/sys/block/mmcblk0/mmcblk0p7/uevent': ''
[... cut ...]
mmcblk0p29      grow               294895
                Total:           61018348

Saving backup images in ./twrp-backup-2016-06-05--19-34-28/ ...
Saving partition boot (mmcblk0p7), 16 MiB uncompressed...
Traceback (most recent call last):
  File "tetherback.py", line 230, in <module>
    if not really_umount('/dev/block/'+devname, mount):
  File "tetherback.py", line 123, in really_umount
    mdev, mnode = (f[0], f[2]) if (f[1], f[3])==('on','type') else (f[0], f[1])
IndexError: list index out of range
`

dlenski, 2016-06-06:
Thanks for the bug report, both of you!

It seems that the file format in your devices' `/sys` folders is different
from what tetherback is expecting.

Could please `adb shell` to the phone while it's booted into TWRP recover,
and then do `cat /sys/block/mmcblk0/uevent`, and show me the complete
contents?

Also, what is the kernel reported by `uname -a` (on the device)?

dlenski, 2016-06-06:
Actually, now that I look at the logs more carefully…

What is the output of `mount ` on the device? (Again in recovery.) Seems like its output is unexpected as well.

Arno500, 2016-06-06:
Here is the cat:
```
adb shell
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
~ # cat /sys/block/mmcblk0/uevent
MAJOR=179
MINOR=0
DEVNAME=mmcblk0
DEVTYPE=disk
NPARTS=29
~ #
```
And the mount:
```
~ # mount
rootfs on / type rootfs (rw,seclabel)
tmpfs on /dev type tmpfs (rw,seclabel,nosuid,relatime,size=1456556k,nr_inodes=140758,mode=755)
devpts on /dev/pts type devpts (rw,seclabel,relatime,mode=600)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,seclabel,relatime)
selinuxfs on /sys/fs/selinux type selinuxfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,seclabel,relatime,size=1456556k,nr_inodes=140758)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
/dev/block/mmcblk0p28 on /data type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/mmcblk0p28 on /sdcard type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/mmcblk0p16 on /cache type ext4 (rw,seclabel,relatime,data=ordered)
~ #
```

dlenski, 2016-06-06:
Thanks, @Arno500, that's helpful. Hmmm... I'm not a bit confused here.

1. tetherback *is* actually figuring out the correct partition map (although it complains about blank lines)
2. `mount` appears to have the expected output format

I added a few more checks on the output in e3a68151. Does that make any difference, or show any new errors? (Hopefully it gets rid of the warnings about blank lines in `uevent` files.)

Joyomojo, 2016-06-06:
```
C:\Users\Joyo>adb -d shell
~ # cat /sys/block/mmcblk0/uevent
MAJOR=179
MINOR=0
DEVNAME=mmcblk0
DEVTYPE=disk
NPARTS=25
```
```
~ # uname -a
Linux localhost 3.4.0-g83662a8-01427-gad51ea1 #1 SMP PREEMPT Sun Dec 7 18:19:05 CST 2014 armv7l GNU/Linux
```
```
~ # mount
rootfs on / type rootfs (rw)
tmpfs on /dev type tmpfs (rw,seclabel,nosuid,relatime,size=1453928k,nr_inodes=174261,mode=755)
devpts on /dev/pts type devpts (rw,seclabel,relatime,mode=600)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,seclabel,relatime)
selinuxfs on /sys/fs/selinux type selinuxfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,seclabel,relatime,size=1453928k,nr_inodes=174261)
/dev/block/mmcblk0p24 on /cache type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/mmcblk0p25 on /data type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/mmcblk0p25 on /sdcard type ext4 (rw,seclabel,relatime,data=ordered)
```

Updated script gives the following error:
```
D:\Downloads>py tetherback.py
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (25 partitions)...
  partition map: 100% Time: 0:00:00
Saving backup images in .\twrp-backup-2016-06-06--18-05-25/ ...
Saving partition boot (mmcblk0p19), 32 MiB uncompressed...
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
WARNING: don't understand output from mount: b''
  boot.emmc.win: 100% Time: 0:00:02   3.5 MiB/s   7.9 MiB
Saving tarball of mmcblk0p23 (mounted at /system), 1280 MiB uncompressed...
Traceback (most recent call last):
  File "tetherback.py", line 227, in <module>
    fstype = really_mount('/dev/block/'+devname, mount)
  File "tetherback.py", line 111, in really_mount
    mdev, mnode, mtype = (f[0], f[2], f[4]) if (f[1], f[3])==('on','type') else (f[0], f[1], f[2])
IndexError: list index out of range
```
boot.emmc.win (7,93MB) and boot.emmc.win.md5 (49 Bytes) are created this time.

dlenski, 2016-06-07:
I have a hunch that this is an issue with Windows/*ix line ending conversion issues (`adb` is known for changing line endings on Windows in odd ways :-1:). The only way I can imagine that you're getting blank lines from `splitlines()` is if the `\n` line ending is somehow converted to `\n\r` rather than the usual `\r\n`.

What version of `adb` are you using, @Joyomojo?

Anyway, I just added filtering to remove blank lines in a couple more places. See if b6aa670110319e7a6239f940e0ada6d501d7ec61 fixes it?

Joyomojo, 2016-06-07:
I am using version 1.4.3 from [this thread](http://forum.xda-developers.com/showthread.php?t=2588979).

I just tried the new tetherback version and it seems everything is working now. No more errors!
```
D:\Downloads\tetherback-master\tetherback>py tetherback.py
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (25 partitions)...
  partition map: 100% Time: 0:00:00
Saving backup images in .\twrp-backup-2016-06-07--10-16-27/ ...
Saving partition boot (mmcblk0p19), 32 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.7 MiB/s   7.9 MiB
Saving tarball of mmcblk0p23 (mounted at /system), 1280 MiB uncompressed...
  system.ext4.win: 100% Time: 0:00:51   6.5 MiB/s 331.0 MiB
Saving tarball of mmcblk0p25 (mounted at /data), 12979 MiB uncompressed...
  data.ext4.win: 100% Time: 0:05:47   5.7 MiB/s   1.9 GiB

D:\Downloads\tetherback-master\tetherback>
```

Thanks you for the quick fix!

Arno500, 2016-06-07:
Tried it. It now backup Boot, but when it need to do the second partition, it give me this:
```
$ python3 tetherback.py -DRC
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:03
Saving backup images in ./twrp-backup-2016-06-07--11-59-39/ ...
Saving partition boot (mmcblk0p7), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.5 MiB/s  10.1 MiB
Traceback (most recent call last):
  File "tetherback.py", line 303, in <module>
    child.terminate()
  File "/usr/lib/python3.4/subprocess.py", line 1686, in terminate
    self.send_signal(signal.SIGTERM)
  File "/usr/lib/python3.4/subprocess.py", line 1681, in send_signal
    os.kill(self.pid, sig)
ProcessLookupError: [Errno 3] No such process
```

dlenski, 2016-06-07:
@Arno500, are you sure the stackdump file was created by *this* invocation of python? It doesn't seem to have crashed, just thrown an error about the subprocess having already terminated.

Try 34e14c9a2013a580d70307f933f0b1f268a4c46b

Arno500, 2016-06-07:
Effectively, the stackdump could have been a random error on the computer not from the script.
I can report it's working correctly (even perfectly, and doesn't throw any errors !).
Thank you for your fast fixing !!!!
Now ready for a restore tool !

dlenski, 2016-06-07:
Great, glad to hear it!

closed 2016-06-07 by dlenski