dlenski/tetherback #18: userdata problem

theman480, 2016-06-09:
The tool is working perfectly but i cant backup userdata, I'm receiving the following error:
`tetherback: error: Partitions were requested for backup, but not found in the partition map: userdata`

dlenski, 2016-06-09:
@theman480, what model of Android device is this?

On every other device I've seen, the partition mounted at `/data` is actually called `userdata`; on yours, it is called `data` instead.

theman480, 2016-06-09:
Zenfone 2

dlenski, 2016-06-09:
I will have to add a bit of additional code to try to figure out if `/system` or `/data` don't correspond to the expected partition names.

Can you please do `cat /etc/fstab` from an `adb shell` while booted into TWRP recovery? It is possible that TWRP itself has some hints that can help me out here. 

I wonder why some Android device makers didn't follow the standard layout. Rrrggghgh :-1:.

theman480, 2016-06-09:
```
cat /etc/fstab
/dev/block/mmcblk0p18 /system ext4 rw 0 0
/dev/block/mmcblk0p15 /cache ext4 rw 0 0
/dev/block/mmcblk0p14 /config ext4 rw 0 0
/dev/block/mmcblk0p19 /data ext4 rw 0 0
/dev/block/mmcblk0p12 /factory ext4 rw 0 0
/dev/block/mmcblk1p1 /external_sd vfat rw 0 0
 /usb-otg vfat rw 0 0
```
Thank you for your attention!

theman480, 2016-06-10:
@dlenski, any news?

Thank you again!

dlenski, 2016-06-10:
@theman480: The latest version tries harder to figure out the correct partition for `/system`, `/data`, `/cache`. Let me know if it works for you.

closed 2016-06-10 by dlenski

theman480, 2016-06-11:
Works perfectly !! Thank you @dlenski !!

commit ecdac832 (2016-06-10): handle non-standard partition names, as long as the mount points are still right (fixes #18)
commit 8874ecae (2016-06-13): handle non-standard partition names, as long as the mount points are still right (fixes #18)