dlenski/tetherback #19: userdata problem

theman480, 2016-06-11:
@dlenski,  sorry for open the issue again, but now i have the following error:

```
Found ADB version 1.0.32
Device reports kernel 3.10.20-g668a4b4
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (19 partitions)...
  partition map: 100% Time: 0:00:02                                                                                                                                                                    
Saving backup images in ./twrp-backup-2016-06-11--12-27-12/ ...
Saving partition boot (mmcblk0p1), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:00   9.9 MiB/s   9.6 MiB                                                                                                                                              
Saving tarball of mmcblk0p15 (mounted at /cache), 256 MiB uncompressed...
  cache.ext4.win: 100% Time: 0:00:00   1.7 MiB/s 328.5 KiB                                                                                                                                             
Saving tarball of mmcblk0p18 (mounted at /system), 2304 MiB uncompressed...
  system.ext4.win: 100% Time: 0:00:59   8.5 MiB/s 504.3 MiB                                                                                                                                            
Saving tarball of mmcblk0p19 (mounted at /data), 11459 MiB uncompressed...
  data.ext4.win: 100% Time: 0:05:32   7.8 MiB/s   2.5 GiB                                                                                                                                              
Traceback (most recent call last):
  File "/usr/local/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.1.0', 'console_scripts', 'tetherback')()
TypeError: 'module' object is not callable

```

dlenski, 2016-06-11:
It looks like the transfer is successful... I have no idea why it would give an error about the module entry point *after* the transfer.

This might be an issue with the Python installation.

How did you install `tetherback`? More importantly, how did you upgrade to this latest version?

theman480, 2016-06-12:
I removed the old version with: pip3 uninstall tetherback and I installed the new with: pip3 install https://github.com/dlenski/tetherback/archive/master.zip.
Is that right?

dlenski, 2016-06-13:
Yeah, that seems perfectly fine.

Does it give similar errors when it isn't actually backing anything up? Does `tetherback --help` give an error after displaying the command-line options? Does `tetherback -0` give an error after showing the partition table?

dlenski, 2016-06-13:
Let me know if the latest version fixes it. I have a hunch that I did not quite understand the way `entry_points` is supposed to work, and hopefully [this page](https://chriswarrick.com/blog/2014/09/15/python-apps-the-right-way-entry_points-and-scripts/) set me right.

theman480, 2016-06-15:
@dlenski I am sorry for the delay, was traveling.
Updated, now i am stuck at data.ext4.win saving, just stuck at 0%
```
Saving tarball of mmcblk0p19 (mounted at /data), 11459 MiB uncompressed...
data.ext4.win:   0% ETA:  --:--:--   0.0 s/B   0.0 B 
```

dlenski, 2016-06-15:
Which transport method? `--tcp`, `--pipe-xo`, something else?

theman480, 2016-06-15:
I am running the following command:

`tetherback -MDRC`

dlenski, 2016-06-16:
Which version are you using?

What does `git describe --always` show?

Please include the first lines of the program's output, since that includes
the information I need.

theman480, 2016-06-16:
git describe --always:
> 34e14c9

tetherback -MDRC:
`ota@ota-Inspiron-5558:$ tetherback -MDRC
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.20-g668a4b4
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (19 partitions)...
  partition map: 100% Time: 0:00:02                                                                                                                  
Saving backup images in ./twrp-backup-2016-06-15--22-30-25/ ...
Saving backup images in ./twrp-backup-2016-06-15--22-30-25/ ...
Saving partition boot (mmcblk0p1), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:00  10.0 MiB/s   9.6 MiB                                                                                            
Saving partition recovery (mmcblk0p2), 16 MiB uncompressed...
  recovery.emmc.win: 100% Time: 0:00:01  11.7 MiB/s  15.1 MiB                                                                                        
Saving tarball of mmcblk0p15 (mounted at /cache), 256 MiB uncompressed...
  cache.ext4.win: 100% Time: 0:00:00   1.7 MiB/s 328.5 KiB                                                                                           
Saving tarball of mmcblk0p18 (mounted at /system), 2304 MiB uncompressed...
  system.ext4.win: 100% Time: 0:00:59   8.5 MiB/s 504.3 MiB                                                                                          
Saving tarball of mmcblk0p19 (mounted at /data), 11459 MiB uncompressed...
  data.ext4.win:   0% ETA:  --:--:--   0.0 s/B   0.0 B                                                                                               

`

dlenski, 2016-06-16:
Thanks, I changed the recent versions to use the `--exec-out ` transport by default, instead of `--tcp`, since I thought it would work correctly on all OSes. But it seems not to be working on your system :-(

Do you have the same problem if you run this version with `--tcp`?

theman480, 2016-06-17:
Same problem :
```
tetherback -MDRC --tcp
Found ADB version 1.0.32
Device reports kernel 3.10.20-g668a4b4
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (19 partitions)...
  partition map: 100% Time: 0:00:02                                            
Saving backup images in ./twrp-backup-2016-06-16--21-41-39/ ...
Saving backup images in ./twrp-backup-2016-06-16--21-41-39/ ...
Saving partition boot (mmcblk0p1), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:00  12.4 MiB/s   9.6 MiB                      
Saving partition recovery (mmcblk0p2), 16 MiB uncompressed...
  recovery.emmc.win: 100% Time: 0:00:01  12.7 MiB/s  15.1 MiB                  
Saving tarball of mmcblk0p15 (mounted at /cache), 256 MiB uncompressed...
  cache.ext4.win: 100% Time: 0:00:00  10.0 MiB/s 328.5 KiB                     
Saving tarball of mmcblk0p18 (mounted at /system), 2304 MiB uncompressed...
  system.ext4.win: 100% Time: 0:00:59   8.4 MiB/s 504.3 MiB                                                 
Saving tarball of mmcblk0p19 (mounted at /data), 11459 MiB uncompressed...
  data.ext4.win:   0% ETA:  --:--:--   0.0 s/B   0.0 B                                                      

```

dlenski, 2016-06-17:
Hmmm. Which build were you using previously, that did **not** have
this problem? Was it v0.4 (34e14c9)?

This is rather puzzling because it manages to copy the first three
partitions just fine, and then only gets stuck on the last one.

theman480, 2016-06-17:
Yes, my first version was 0.4.

dlenski, 2016-06-17:
And is that version still working correctly for you?

theman480, 2016-06-17:
No, with the first version i can't save the data partition.

dlenski, 2016-06-17:
Ah, right. <smacks forehead>… your original problem.

I think I figured out what was causing this. Some devices try to put temporary files in a location that I didn't expect, which interacts badly with the way I set up the transfer. You can read 2aeca1106b553b760cb32bf16a0f6ab26b0481ca for the gory details. Do let me know if this works for you now!

theman480, 2016-06-18:
It works perfectly!
Thank you @dlenski for your nice support!
Just one question i have a folder inside of the backup folder with the same name but the backup folder has 7.0GB and the folder inside with the same name has 3.5GB, is that right?

dlenski, 2016-06-18:
Glad to hear it.

Take a look inside the backup folder. You'll see that every time it is run, it creates a new sub-folder, with the current date and time.

closed 2016-06-18 by dlenski