dlenski/tetherback #20: OverflowError (utils.py, line 222, function format_time)

pati610, 2016-06-25:
Looks very promising! I get an error, though, "OverflowError: normalized days too large to fit in a C int", issued by the function format_time in utils.py, line 222. Any ideas why?

Patrick
...................

Device reports TWRP version 2.8.6.1
Reading partition map for mmcblk0 (47 partitions)...
  partition map: 100% Time: 0:00:07                                            
Saving backup images in ./twrp-backup-2016-06-25--06-10-28/ ...
Saving partition boot (mmcblk0p42), 16 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.4 MiB/s   7.5 MiB                      
Saving tarball of mmcblk0p45 (mounted at /system), 2688 MiB uncompressed...
Traceback (most recent call last):
  File "tetherback.py", line 255, in <module>
    pbar = ProgressBar(max_value=size*512, widgets=pbwidgets).start()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 539, in start
    self.update(self.min_value, force=True)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 496, in update
    StdRedirectMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 131, in update
    DefaultFdMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 42, in update
    self.fd.write('\r' + self._format_line())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 433, in _format_line
    widgets = ''.join(self._format_widgets())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 413, in _format_widgets
    widget_output = widget(self, data)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/widgets.py", line 289, in __call__
    data['eta'] = utils.format_time(data['eta_seconds'])
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/utils.py", line 222, in format_time
    time = datetime.timedelta(seconds=int(time))
OverflowError: normalized days too large to fit in a C int
.................

dlenski, 2016-06-25:
Hmmm... I'm not sure what to make of this one. The error is apparently occurring inside the `progressbar` library, upon calling the `start` function more specifically.

Couple questions …

1. What version of the `progressbar` package do you have, and how did you install it?
2. What options are you running `tetherback` with? Please post the partition table of your phone (included with `-v`) so we can check that it looks reasonable.

dlenski, 2016-06-25:
You must also be running an old version of `tetherback`, since the first output lines don't match the current version. Please include the version you're running.

dlenski, 2016-06-25:
My guess would be that you're running `v0.3`, since that's the example I showed in the README… I just tried running that build and don't get any errors myself.

pati610, 2016-06-26:
Hi Dan, and thank so much for your reply!

I've tested every single version of tetherback, from 0.3 ->. They are all listed by pip3 as tetherback (0.1.0), though, is that normal?

Here is the pip3 list:
...............
pati$ pip3 list
pip (8.1.2)
progressbar2 (3.9.1)
python-utils (2.0.0)
setuptools (18.2)
six (1.10.0)
tabulate (0.7.5)
tetherback (0.1.0)
................

And here is the output of tetherback -v, running v0.6:

.............
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-g973caf2
Device reports TWRP version 2.8.6.1
Reading partition map for mmcblk0 (47 partitions)...
  partition map: 100% Time: 0:00:02                                            

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       sbl1                      1024
mmcblk0p2       pg1fs                   121823
mmcblk0p3       board_info                  16
mmcblk0p4       reserve_1                 8192
mmcblk0p5       mfg                        256
mmcblk0p6       pg2fs                    24255
mmcblk0p7       sbl1_update                256
mmcblk0p8       rpm                        256
mmcblk0p9       tz                        2048
mmcblk0p10      sdi                         32
mmcblk0p11      hboot                     4096
mmcblk0p12      sp1                       5120
mmcblk0p13      wifi                      1024
mmcblk0p14      ddr                         32
mmcblk0p15      dsps                      1024
mmcblk0p16      adsp                     61441
mmcblk0p17      wcnss                     5120
mmcblk0p18      radio_config              8192
mmcblk0p19      fsg                       1536
mmcblk0p20      radio                    76801
mmcblk0p21      tool_diag                 4096
mmcblk0p22      custdata                 51200
mmcblk0p23      reserve_2                15359
mmcblk0p24      misc                      1024
mmcblk0p25      modem_st1                 1536
mmcblk0p26      modem_st2                 1536
mmcblk0p27      fataldevlog              20480
mmcblk0p28      debug_config                 4
mmcblk0p29      pdata                      256
mmcblk0p30      control                     16
mmcblk0p31      local                     1281
mmcblk0p32      extra                       64
mmcblk0p33      cdma_record               1024
mmcblk0p34      fsc                          1
mmcblk0p35      ssd                          8
mmcblk0p36      skylink                    256
mmcblk0p37      carrier                  25600
mmcblk0p38      sensor_hub                 256
mmcblk0p39      devlog                   30720  /devlog        ext4
mmcblk0p40      cir_img                     10
[... 31 lines cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/__main__.py", line 3, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/tetherback.py", line 308, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/tetherback.py", line 247, in backup_partition
    pbar = ProgressBar(max_value=pi.size*512, widgets=pbwidgets).start()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 539, in start
    self.update(self.min_value, force=True)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 496, in update
    StdRedirectMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 131, in update
    DefaultFdMixin.update(self, value=value)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 42, in update
    self.fd.write('\r' + self._format_line())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 433, in _format_line
    widgets = ''.join(self._format_widgets())
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/bar.py", line 413, in _format_widgets
    widget_output = widget(self, data)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/widgets.py", line 289, in __call__
    data['eta'] = utils.format_time(data['eta_seconds'])
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/progressbar/utils.py", line 222, in format_time
    time = datetime.timedelta(seconds=int(time))
OverflowError: normalized days too large to fit in a C int

dlenski, 2016-06-26:
Thanks for filling in more details! (BTW, please put \`\`\` before and after blocks of code or logs to be included literally, for ease of reading; I added this to your post)

Your device has a much larger number of partitions than any other Android device I've seen. But I don't think that's got anything to do with the problem.

The issue seems to be in `progressbar2` itself; apparently, it is having trouble with some time-based computation. It's especially puzzling because…

* I noticed that you're using progressbar2 v3.9.1, whereas I have been using v3.6.2. I upgraded to v3.9.1 myself, to see if I would have the same problem, but I don't.
* The same progressbar widget (`ETA`) is used for generating the partition map and for the `boot` partition. Apparently there is no problem until you get to the `/system` partition.

Could you lightly modify tetherback to not include the `ETA` widget in the progress bar, and let me know if it works without this?

```diff
diff --git a/tetherback/tetherback.py b/tetherback/tetherback.py
index 73debf5..40e9fbf 100755
--- a/tetherback/tetherback.py
+++ b/tetherback/tetherback.py
@@ -245,7 +245,7 @@ def backup_partition(adb, pi, bp, transport, verify=True):
         s.connect(('localhost', port))
         block_iter = iter(lambda: s.recv(65536), b'')
 
-    pbwidgets = ['  %s: ' % bp.fn, Percentage(), ' ', ETA(), ' ', FileTransferSpeed(), ' ', DataSize() ]
+    pbwidgets = ['  %s: ' % bp.fn, Percentage(), ' ', FileTransferSpeed(), ' ', DataSize() ]
     pbar = ProgressBar(max_value=pi.size*512, widgets=pbwidgets).start()
 
     with open(bp.fn, 'wb') as out:
```

pati610, 2016-06-27:
Thanks for the heads-up regarding proper syntax here :) I did the modification, and now I get this error:

> Patis-iMac:scripts pati$ tetherback -v
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-g973caf2
Device reports TWRP version 2.8.6.1
Reading partition map for mmcblk0 (47 partitions)...
  partition map: 100% Time: 0:00:07                                            

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       sbl1                      1024
mmcblk0p2       pg1fs                   121823
mmcblk0p3       board_info                  16
mmcblk0p4       reserve_1                 8192
mmcblk0p5       mfg                        256
mmcblk0p6       pg2fs                    24255
mmcblk0p7       sbl1_update                256
mmcblk0p8       rpm                        256
mmcblk0p9       tz                        2048
mmcblk0p10      sdi                         32
mmcblk0p11      hboot                     4096
mmcblk0p12      sp1                       5120
mmcblk0p13      wifi                      1024
mmcblk0p14      ddr                         32
mmcblk0p15      dsps                      1024
mmcblk0p16      adsp                     61441
mmcblk0p17      wcnss                     5120
mmcblk0p18      radio_config              8192
mmcblk0p19      fsg                       1536
mmcblk0p20      radio                    76801
mmcblk0p21      tool_diag                 4096
mmcblk0p22      custdata                 51200
mmcblk0p23      reserve_2                15359
mmcblk0p24      misc                      1024
mmcblk0p25      modem_st1                 1536
mmcblk0p26      modem_st2                 1536
mmcblk0p27      fataldevlog              20480
mmcblk0p28      debug_config                 4
mmcblk0p29      pdata                      256
mmcblk0p30      control                     16
mmcblk0p31      local                     1281
mmcblk0p32      extra                       64
mmcblk0p33      cdma_record               1024
mmcblk0p34      fsc                          1
mmcblk0p35      ssd                          8
mmcblk0p36      skylink                    256
mmcblk0p37      carrier                  25600
mmcblk0p38      sensor_hub                 256
mmcblk0p39      devlog                   30720
mmcblk0p40      cir_img                     10
mmcblk0p41      reserve                  47000
mmcblk0p42      boot                     16384
mmcblk0p43      recovery                 24576
mmcblk0p44      reserve_3                90112
mmcblk0p45      system                 2752512
mmcblk0p46      cache                   393216
mmcblk0p47      userdata              11403264
                Total:                15204335

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  -------------------------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="media*" --exclude="*-cache"
[... 22 lines cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2682, in load_entry_point
    return ep.load()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2355, in load
    return self.resolve()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/__main__.py", line 3, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/tetherback.py", line 308, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/tetherback.py", line 260, in backup_partition
    devicemd5 = adb.check_output(('shell','cat %s/md5out' % tmpdir)).strip().split()[0]
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/tetherback/adb_wrapper.py", line 30, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 629, in check_output
    **kwargs).stdout
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 698, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 1055, in communicate
    stdout = self.stdout.read()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte

>

dlenski, 2016-06-28:
Looks like we're making progress… I'm fuzzy on why you're getting a `UnicodeDecodeError` specifically, but I think you're now running into issue #19.

That bug was introduced in v0.6~3 (by my too-clever-by-half attempt to select a random temporary directory) and fixed in v0.6.1.

(I still need to address your point about the version number in `setup.py` not matching the git tags. :-1:)

pati610, 2016-07-01:
Hi again Dan,

and thank you for all your help! I think it all might have come down to the sorry state of my M8 - it got smashed (or crushed might be the more appropriate description) by a closing car door, and I tried everything I could to get the data recovered. I think there might have been serious problems with the device (besides its not-too-attractive appearance) which might have triggered the aforementioned errors here.

I got my data back using a palette of different methods. Then I went to buy a new, similar device - I thought. Didn't notice the "s" after the M8, and the Htc One M8s is a very different phone. 

Now, needless to say, I ran into a multitude of problems getting custom recovery + root on this new device. It took me the better part of two days (of which the first day was in vain, trying to use the M8 procedure I was accustomed with). But as of now I have it, TWRP, root and busybox (*phew*).

Using tetherback on the new device, I was able to backup boot and system without any problems! But:

As the internal storage is kind of small for todays needs (16 GB), I opted for the adoptable storage option, which basically turns the ext-microsd into an internal drive. This is great for space, but introduced new problems.

None of the TWRP img's for M8S are as of now able to handle the adoptable storage (in contrast to the newer releases for other devices). My TWRP reads only the internal flash drive, and lists ext-microsd as 0 b.

This is just throwing it out there, but do you think it would be possible to modify tetherback to handle partitions on mmcblk1 as well? I know some Python, and could give it a go if you think it's not a doomed idea from the start. The userdata and apps are as of 6.0 is split between the original mount and the new mount (mmcblk1p2).

I'd be so much more eager to try a solution like this than trying to compile a custom TWRP for my device - I don't have any experience in that field at all.

As of now, I'm able to backup boot and system through TWRP and tetherback, and then backup the rest by adb admin root + insecure adbd, pulling the files at the mount points for the card. It works, but it's complicated and slow.

Any input here would be of great value.

Regards, 
Patrick

dlenski, 2016-07-01:
I'm glad to hear it's been of some use!

> would be possible to modify tetherback to handle partitions on mmcblk1 as well

Definitely. This would be an easy addition. I've never seen an Android device that has multiple `mmcblk` devices, so I wouldn't be able to test it myself, but I have no reason to think it'd be any different from `mmcblk1`.

pati610, 2016-07-01:
Thanks Dan, great news - I'll give it a shot!

closed 2016-07-01 by dlenski

pati610, 2016-07-01:
Changing the value for mmcblk in build_partmap lists the mmcblk1 without problems:

'''
$ tetherback -0
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.49-gf9241cc
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk1 (2 partitions)...
  partition map: 100% Time: 0:00:00                                            

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk1p1       android_meta             16384  /external_sd   vfat
mmcblk1p2       android_expand        62350319  /sd-ext        ext4
                Total:                62366703
'''

What would be your take on this - a for loop thrown into build_partmap? And using the already existing -X flag to backup android_meta and android_expand?

dlenski, 2016-07-03:
> What would be your take on this... a for loop thrown into build_partmap? And using the already existing -X flag to backup android_meta and android_expand? Or running the function as is, twice (with an option of omitting cache, system, data, boot)?

As you've seen, what I did in [`feature/multi_mmcblk`](https://github.com/dlenski/tetherback/tree/feature/multi_mmcblk) was simply to loop over all the `mmcblk*` devices.

(You can already omit the default partitions with `--no-boot`, `--no-system`, `--no-userdata`.)