dlenski/tetherback #21: Data corruption with stty -onlcr

Phaeilo, 2016-07-02:
Hello,
I'm not actually using tetherback, but I have been backing up my Nexus 4 with my own shell script using your `stty -onlcr` method. I am running TWRP 3.0.2-0 and ADB 1.0.32. While restoring from my backups, I noticed that they were corrupted. After thoroughly debugging my backup process, it seems as if a few bytes get dropped or altered during the transfer. I wanted to share these results with you:

```
% adb version
Android Debug Bridge version 1.0.32
Revision 6.0.1_r46

% adb shell "dd if=/dev/block/platform/msm_sdcc.1/by-name/system | md5sum"
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
1720320+0 records in
1720320+0 records out
880803840 bytes (840.0MB) copied, 22.685457 seconds, 37.0MB/s
d8f9a197c857a8b532a8aed63550e390  -

% for x in $(seq 10) ; do adb shell "stty -onlcr && dd if=/dev/block/platform/msm_sdcc.1/by-name/system 2>/dev/null" > backup_$x ; done

% md5sum backup_*
d8f9a197c857a8b532a8aed63550e390  backup_1
890c7c82efe3f2813cc4e39cc14b79f4  backup_10
d8f9a197c857a8b532a8aed63550e390  backup_2
d8f9a197c857a8b532a8aed63550e390  backup_3
c79a6bafe3327963d8259b6f17948dac  backup_4
95c93196448f3895669c074861706d45  backup_5
d6785fc7016c1a91f004f938ec44e3ae  backup_6
d8f9a197c857a8b532a8aed63550e390  backup_7
d8f9a197c857a8b532a8aed63550e390  backup_8
7ee6cfe40f09bb80da8c8f7378466e14  backup_9

% adb shell "dd if=/dev/block/platform/msm_sdcc.1/by-name/system | md5sum"                                                             
1720320+0 records in
1720320+0 records out
880803840 bytes (840.0MB) copied, 22.695071 seconds, 37.0MB/s
d8f9a197c857a8b532a8aed63550e390  -
```

dlenski, 2016-07-02:
What host platform are you running on?

It appears to be well-known at this point that this technique for getting an 8-bit-clean pipe does not work on Windows: http://stackoverflow.com/a/20141481/20789

Since you are using adb v1.0.32, you should use `adb exec-out` instead. See the comments in [tetherback.py#L220-L246](https://github.com/dlenski/tetherback/blob/master/tetherback/tetherback.py#L220-L246) for some more details on the possible transfer methods.

As far as I can tell, the `stty -onlcr` trick **only** works on Linux. Because of the potential for corruption, I added the md5sum checking to tetherback.

Phaeilo, 2016-07-03:
HI @dlenski, I'm executing these commands in ZSH on ArchLinux.

dlenski, 2016-07-03:
When you run a binary diff between the different versions of the `/system ` partition images, what do you see?

Phaeilo, 2016-07-03:
I have attached a screenshot of the binary diff of backup_1 and backup_4. The good backup_1 is shown on the top, the corrupted backup_4 on the bottom. You can see that there is quite a considerable chunk of bytes missing from backup_4. I have highlighted these bytes in backup_1.

[image]

dlenski, 2016-07-03:
Very, very odd. The only corruption I've seen with `adb shell` has been the unwanted CR/LF translation.

If you rerun this same script but changing `adb shell` to `adb exec-out`, do you get the same corruption?

Phaeilo, 2016-07-03:
I don't see any corruption with `adb exec-out`.

```
% adb version
Android Debug Bridge version 1.0.32
Revision 6.0.1_r46

% adb shell "dd if=/dev/block/platform/msm_sdcc.1/by-name/system | md5sum"
* daemon not running. starting it now on port 5037 *
* daemon started successfully *
1720320+0 records in
1720320+0 records out
27928a6015529d13bb1564c39e042fab  -
880803840 bytes (840.0MB) copied, 22.663940 seconds, 37.1MB/s

% for x in $(seq 10) ; do adb exec-out "dd if=/dev/block/platform/msm_sdcc.1/by-name/system 2>/dev/null" > execout_$x ; done

% md5sum execout_*
27928a6015529d13bb1564c39e042fab  execout_1
27928a6015529d13bb1564c39e042fab  execout_10
27928a6015529d13bb1564c39e042fab  execout_2
27928a6015529d13bb1564c39e042fab  execout_3
27928a6015529d13bb1564c39e042fab  execout_4
27928a6015529d13bb1564c39e042fab  execout_5
27928a6015529d13bb1564c39e042fab  execout_6
27928a6015529d13bb1564c39e042fab  execout_7
27928a6015529d13bb1564c39e042fab  execout_8
27928a6015529d13bb1564c39e042fab  execout_9
md5sum execout_*  11,80s user 1,69s system 99% cpu 13,494 total

% adb shell "dd if=/dev/block/platform/msm_sdcc.1/by-name/system | md5sum"
1720320+0 records in
1720320+0 records out
880803840 bytes (840.0MB) copied, 22.834518 seconds, 36.8MB/s
27928a6015529d13bb1564c39e042fab  -
```

dlenski, 2016-07-03:
Thanks. These are all very interesting and useful datapoints.

It's hard to know that root cause with so much variation in host and device versions of `adb`, but my conclusion is that `adb shell` is unreliable, even with the `stty -onlcr` trick and even under Linux.

This makes me feel pretty good about having chosen `exec-out` as the default transport with adb >1.0.32, and TCP with earlier versions:

```
def sensible_transport(transport, adbversion):
    if transport==adbxp.pipe_xo and adbversion<(1,0,32):
        print("WARNING: exec-out pipe (--exec-out) won't work with ADB version < 1.0.32; changing to TCP", file=stderr)
        return adbxp.tcp
    elif transport==adbxp.pipe_bin:
        if adbversion>=(1,0,32):
            print("WARNING: adb shell pipe (--pipe) not needed with ADB >= 1.0.32; changing to adb exec-out pipe", file=stderr)
            return adbxp.pipe_xo
        elif sys.platform.startswith('linux'):
            print("WARNING: adb shell pipe (--pipe) only works on Linux host; changing to TCP", file=stderr)
            return adbxp.tcp
    elif transport is None:
        if adbversion>=(1,0,32):
            print("Using default transfer method: adb exec-out pipe (--exec-out)", file=stderr)
            return adbxp.pipe_xo
        else:
            print("Using default transfer method: adb TCP forwarding (--tcp)", file=stderr)
            return adbxp.tcp
    return transport
```

closed 2016-07-03 by dlenski