dlenski/tetherback #22: Error with LG G4 H815

Fischman132, 2016-07-02:
When trying run trough ADB binary pipe following happens:

`tim@Tim-FISH970A:/media/tim/Dateien/LG G4/backtoStock/Backups$ ./tetherback.py -R
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (51 partitions)...
  partition map: 100% Time: 0:00:08                                            
Saving backup images in ./twrp-backup-2016-07-02--20-26-46/ ...
Saving partition boot (mmcblk0p38), 40 MiB uncompressed...
  boot.emmc.win:   0% ETA:  --:--:--   0.0 s/B                                 Traceback (most recent call last):
  File "./tetherback.py", line 254, in <module>
    pbar.update(out.tell())
  File "/usr/local/lib/python3.5/dist-packages/progressbar/bar.py", line 475, in update
    StdRedirectMixin.update(self, value=value)
  File "/usr/local/lib/python3.5/dist-packages/progressbar/bar.py", line 110, in update
    DefaultFdMixin.update(self, value=value)
  File "/usr/local/lib/python3.5/dist-packages/progressbar/bar.py", line 42, in update
    self.fd.write('\r' + self._format_line())
  File "/usr/local/lib/python3.5/dist-packages/progressbar/bar.py", line 412, in _format_line
    widgets = ''.join(self._format_widgets())
  File "/usr/local/lib/python3.5/dist-packages/progressbar/bar.py", line 392, in _format_widgets
    widget_output = widget(self, data)
  File "/usr/local/lib/python3.5/dist-packages/progressbar/widgets.py", line 282, in __call__
    data['eta'] = utils.format_time(data['eta_seconds'])
  File "/usr/local/lib/python3.5/dist-packages/progressbar/utils.py", line 228, in format_time
    time = datetime.timedelta(seconds=int(time))
OverflowError: days=2127920522; must have magnitude <= 999999999`

Fischman132, 2016-07-02:
Pastebin:http://pastebin.com/VQHxaSzv

dlenski, 2016-07-02:
This looks extremely similar to #20.

What version of `progressbar` are you using?

Does removing the `ETA` widget from the Progress Bar solve the problem, as I suggested in [a comment on #20](https://github.com/dlenski/tetherback/issues/20#issuecomment-228609974)?

(I am beginning to think this is a bug in `progressbar`, not in `tetherback`.)

dlenski, 2016-07-02:
Also, it appears from your log that you're running an older version of tetherback.

Please try the latest release, v0.6.1: https://github.com/dlenski/tetherback/releases/latest

dlenski, 2016-07-03:
I'm now fairly certain this is a bug in progressbar, which was fixed in [`cf19031`](https://github.com/WoLpH/python-progressbar/commit/cf19031aad1d4474d995aa518834382169d2b758 ) but not yet released.

Downgrading to progressbar v3.6.1 works for now.

closed 2016-07-03 by dlenski