dlenski/tetherback #23: Possibility of restoring from OTG?

pati610, 2016-07-05:
Hi again Dan,

and thank you so much for you modding of your original script into feature/tetherback_extra_mount. I'm halfway through testing, and so far it works flawlessly! 

Since the raw img of android_expand is going to be HUGE, I'm pondering the possibility to restore from OTG. Do you think it would be a viable option? My version of TWRP doesn't "see" the ext-card used as internal storage (introduced since 6.0). Thus, it's impossible for me to get the large image back to my combined internal + micro-sd memory.

My rudimental TWRP (the M8s is kind of lagging behind) recognizes OTG memory. Do you think the TWRP would be able to restore a raw image of the internally-configured ext card?

Another great thing would naturally be a "reversed tetherback", as mentioned in other threads. I saw that you've been working some on that. I do too, over here. If I get something going, I'll get back to you on that.

Thanks again, I really appreciate your kind effort.

dlenski, 2016-07-05:
Hmmm. It seems like TWRP *should* be able to restore from OTG if it can see it, but unfortunately I don't have any way to test this myself since I don't have an OTG storage device handy.

The "right" solution from my point of view would be for me to actually get around to implementing an automated `tetherrestore`, as requested back in #5.

dlenski, 2016-07-05:
Patches/pull requests to add `tetherrestore` are always welcome :-D