dlenski/tetherback #24: Issue on Nexus 5

fladson, 2016-08-04:
First, thanks for the awesome tool, appreciate it.
Just reporting an issue that could help to improve it.

Device: Nexus 5 - Android 6.0.1 (TWRP v2.8.7.1)

```
$ tetherback                                                                                                                                           
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-bricked-hammerhead-twrp-g7b77eb4
Device reports TWRP version 2.8.7.1
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100% Time: 0:00:01
Saving backup images in ./twrp-backup-2016-08-04--14-24-19/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:02   3.6 MiB/s   7.9 MiB
Saving tarball of mmcblk0p25 (mounted at /system), 1024 MiB uncompressed...
  system.ext4.win: 100% Time: 0:01:31   5.8 MiB/s 530.7 MiB
Saving tarball of mmcblk0p28 (mounted at /data), 13089 MiB uncompressed...
  data.ext4.win: 100% Time: 0:11:07   3.0 MiB/s   2.0 GiB
error: no devices found
Traceback (most recent call last):
  File "/usr/local/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.6.3', 'console_scripts', 'tetherback')()
  File "/usr/local/lib/python3.5/site-packages/tetherback/tetherback.py", line 309, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/usr/local/lib/python3.5/site-packages/tetherback/tetherback.py", line 261, in backup_partition
    devicemd5 = adb.check_output(('shell','cat /tmp/md5out && rm -f /tmp/md5in /tmp/md5out')).strip().split()[0]
  File "/usr/local/lib/python3.5/site-packages/tetherback/adb_wrapper.py", line 30, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "/usr/local/Cellar/python3/3.5.2_1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/local/Cellar/python3/3.5.2_1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '('adb', '-d', 'shell', 'cat /tmp/md5out && rm -f /tmp/md5in /tmp/md5out')' returned non-zero exit status 1
```

fladson, 2016-08-04:
I'm trying with `-N` now to see the result.

dlenski, 2016-08-04:
Hi @fladson... the first one appears to be showing that your device became disconnected from `adb`. Doesn't seem to be an issue with tetherback per se. Or am I missing something?

Can you try running with `-v` (verbose) and let me know the output?

dlenski, 2016-08-10:
I think figured out what was causing this: when a backup is interrupted but the device isn't interrupted, the copying process (`dd` or `tar`) hangs around on the device, which is undesirable.

27913d0 should at least partially mitigate this by trying to do some cleanup in the case of interruption/exception.

closed 2016-10-29 by dlenski