dlenski/tetherback #26: File not found?

Kotosif, 2016-08-12:
Tried to do the TWRP style backup and this is what happens

```
C:\WINDOWS\system32>tetherback
tetherback-script.py v0.7
Found ADB version 1.0.31
Using default transfer method: adb TCP forwarding (--tcp)
Traceback (most recent call last):
  File "D:\Python34\Scripts\tetherback-script.py", line 9, in <module>
    load_entry_point('tetherback==0.7', 'console_scripts', 'tetherback')()
  File "d:\python34\lib\site-packages\tetherback\tetherback.py", line 311, in main
    check_TWRP(p, adb, args.force)
  File "d:\python34\lib\site-packages\tetherback\tetherback.py", line 83, in check_TWRP
    kernel = adb.check_output(('shell','uname -r')).strip()
  File "d:\python34\lib\site-packages\tetherback\adb_wrapper.py", line 30, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "d:\python34\lib\subprocess.py", line 607, in check_output
    with Popen(*popenargs, stdout=PIPE, **kwargs) as process:
  File "d:\python34\lib\subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "d:\python34\lib\subprocess.py", line 1114, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

dlenski, 2016-08-12:
Where is your `adb` binary located?

I'm a bit confused because the ADB binary was located correctly for the version check, but then it's not found in a subsequent command.

I've never used Windows, so I'm not sure if there's anything special about invoking `adb` on it.

Kotosif, 2016-08-12:
It's located in my downloads folder
The whole path is: D:\Users\D****\Downloads\sdk adt-bundle-windows-x86_64-20130522\sdk\platform-tools\adb.exe

dlenski, 2016-08-21:
Could you try the latest version ([v0.8](https://github.com/dlenski/tetherback/releases/v0.8)) as `tetherback -v -v`, which will show each individual `adb` command as it's executed?

Kotosif, 2016-08-21:
Okay here is the cmd output:

```
C:\WINDOWS\system32>tetherback -v -v
tetherback-script.py v0.8
ADB: ('adb', '-d', 'version')
Found ADB version 1.0.31
Using default transfer method: adb TCP forwarding (--tcp)
ADB: ('adb', '-d', 'shell', 'uname -r')
Traceback (most recent call last):
  File "d:\python34\scripts\tetherback-script.py", line 9, in <module>
    load_entry_point('tetherback==0.8', 'console_scripts', 'tetherback')()
  File "d:\python34\lib\site-packages\tetherback\tetherback.py", line 299, in main
    check_TWRP(p, adb, args.force)
  File "d:\python34\lib\site-packages\tetherback\tetherback.py", line 84, in check_TWRP
    kernel = adb.check_output(('shell','uname -r')).strip()
  File "d:\python34\lib\site-packages\tetherback\adb_wrapper.py", line 35, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "d:\python34\lib\subprocess.py", line 607, in check_output
    with Popen(*popenargs, stdout=PIPE, **kwargs) as process:
  File "d:\python34\lib\subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "d:\python34\lib\subprocess.py", line 1114, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

dlenski, 2016-08-21:
I am stumped. There must be something about finding executable files on Windows that I don't understand.

Maybe try changing this line in tetherback.py …

```
adb = AdbWrapper('adb', ...
```

… to specify the full absolute path of adb.exe?

```
adb = AdbWrapper(r'c:\whatever\path\to\adb.exe', ...
```

dlenski, 2016-08-21:
The other possibility is that `adb` is not actually in a directory listed in your PATH environment variable, but is in the current directory.

That's probably what's happening here: when tetherback changes into the backup directory, it can no longer find the adb executable which was in the same directory.

Kotosif, 2016-08-21:
I think I managed to fix it. It seems it couldn't cope with the random string of numbers in my pathfile to the adb.exe as when I renamed the folder it worked.

commit fcf28c33 (2016-08-22): don't chdir; should fix #26 (where adb.exe is in the current directory but not in PATH)
commit fc4905a8 (2016-09-18): fixup! don't chdir; should fix #26 (where adb.exe is in the current directory but not in PATH)
closed 2016-08-21 by Kotosif