dlenski/tetherback #27: Python: "can not perform relative import"

kerlerm, 2016-08-21:
Hi Dan!

Latest tetherback fails with the following error:
```
$ tetherback/tetherback.py 
Traceback (most recent call last):
  File "tetherback/tetherback.py", line 18, in <module>
    from .version import __version__
SystemError: Parent module '' not loaded, cannot perform relative import
```

As I recently updated my system to the latest Ubuntu LTS I'm not sure what to blame...

Tetherback is executed directly in the local git repository but also installed as release 0.8 by pip3. Does this interfere?

dlenski, 2016-08-21:
This is a side effect of tetherback being installed as a real Python package now.

You can't invoke it in this way (`tetherback/tetherback.py`) because it doesn't know what package it's part of when you do this.

Easy workaround is to use `pip3 -e` to install it in "development mode" which means that it installs via links rather than copies, so you can continue to edit it and have the changes reflected when you simply run `tetherback`.

dlenski, 2016-08-21:
You can also simply use `python -m tetherback` from the parent dir (this simply means to use `tetherback.__main__` as the entry point).

kerlerm, 2016-08-21:
Thanks for your fast & precise answer. Helpful as always :-)

closed 2016-08-21 by kerlerm