dlenski/tetherback #28: backup of userdata fails: cloud not mount None

kerlerm, 2016-08-21:
When backing up maguro with the latest version, tetherback fails to backup the userdata partition:

```
python3 -m tetherback -x -R -M -v -o /home/nas...
__main__.py v0.7.dev20+ge5a26a4_dirty
Found ADB version 1.0.36
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100%                                                                                                                     
Reading partition map for mmcblk0boot0 (0 partitions)...
  partition map: 100%                                                                                                                     
Reading partition map for mmcblk0boot1 (0 partitions)...
  partition map: 100%                                                                                                                     

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       xloader                    128
mmcblk0p2       sbl                       3584
mmcblk0p3       efs                      20480
mmcblk0p4       param                     8192
mmcblk0p5       misc                      4096
mmcblk0p6       dgs                       4096
mmcblk0p7       boot                      8192
mmcblk0p8       recovery                 12224
mmcblk0p9       radio                    16384
mmcblk0p10      system                  669696  /system        ext4
mmcblk0p11      cache                   442368  /cache         ext4
mmcblk0p12      userdata              14198767
mmcblk0p13      metadata                    64
                Total:                15388271

Backup plan:

PARTITION NAME    FILENAME           FORMAT
----------------  -----------------  ------------------------------
boot              boot.emmc.win      gzipped raw image
recovery          recovery.emmc.win  gzipped raw image
system            system.ext4.win    tar -cz -p
userdata          data.ext4.win      tar -cz -p --exclude="*-cache"

Saving backup images in /home/nas/.../twrp-backup-2016-08-21--19-30-21/ ...
Saving partition boot (mmcblk0p7), 8 MiB uncompressed...
  boot.emmc.win: 100%   2.7 MiB/s   4.9 MiB                                                                                               
Saving partition recovery (mmcblk0p8), 11 MiB uncompressed...
  recovery.emmc.win: 100%   2.9 MiB/s   9.3 MiB                                                                                           
Saving tarball of mmcblk0p10 (mounted at /system), 654 MiB uncompressed...
  system.ext4.win: 100%   2.1 MiB/s 288.0 MiB                                                                                             
Saving tarball of mmcblk0p12 (mounted at None), 13865 MiB uncompressed...
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/src/git_repo/tetherback/tetherback/__main__.py", line 4, in <module>
    main()
  File "/usr/local/src/git_repo/tetherback/tetherback/tetherback.py", line 327, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/usr/local/src/git_repo/tetherback/tetherback/tetherback.py", line 219, in backup_partition
    raise RuntimeError('%s: could not mount %s' % (pi.partname, pi.mountpoint))
RuntimeError: userdata: could not mount None
```

kerlerm, 2016-08-21:
-v -v shows the following:

```
 [...]
Saving tarball of mmcblk0p12 (mounted at None), 13865 MiB uncompressed...
ADB: ('adb', '-d', 'shell', 'mount -o ro /dev/block/mmcblk0p12 None 2>/dev/null && echo ok')
ADB: ('adb', '-d', 'shell', 'mount -o remount,ro /dev/block/mmcblk0p12 None 2>/dev/null && echo ok')
ADB: ('adb', '-d', 'shell', 'mount')
Traceback (most recent call last):
[...]

```

dlenski, 2016-08-21:
For some reason it is not finding the mount point for the userdata partition (notice that the mount point is blank in the partition map).

Does `adb shell cat /etc/fstab` show it?

I did not change the partition finding logic at all in recent versions.

I'm a bit leery of the way you're invoking tetherback (calling the __main__.py directly from the command line. Depending on your PYTHONPATH, you may actually be pulling in submodules from a different version of tetherback installed in another location.

kerlerm, 2016-08-22:
This is the output of cat fstab:
```
~ # cat /etc/fstab 
/dev/block/mmcblk0p10 /system ext4 rw 0 0
/dev/block/mmcblk0p11 /cache ext4 rw 0 0
/dev/block/dm-0 /data ext4 rw 0 0
 /usb-otg vfat rw 0 0
```
I run on an encrypted data partition.

dlenski, 2016-08-22:
Ah, that's the problem. Tetherback only knows how to handle `/dev/block/mmcblk*` positions that are mounted directly, rather than through devicemapper (which is used for encrypted positions).

As an initial band-aid, I should add a check for this case and a more comprehensible error message.

samesamebut, 2016-10-18:
Same here, also with encrypted data partition:

`# tetherback -0
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100% Time: 0:00:00                                                                                                                                                               

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       xloader                    128
mmcblk0p2       sbl                       3584
mmcblk0p3       efs                      20480
mmcblk0p4       param                     8192
mmcblk0p5       misc                      4096
mmcblk0p6       dgs                       4096
mmcblk0p7       boot                      8192
mmcblk0p8       recovery                 12224
mmcblk0p9       radio                    16384
mmcblk0p10      system                  669696  /system        ext4
mmcblk0p11      cache                   442368  /cache         ext4
mmcblk0p12      userdata              14198767
mmcblk0p13      metadata                    64
                Total:                15388271

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  -------------------------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="media*" --exclude="*-cache"
`

and when I run I get following output:

`# tetherback
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100% Time: 0:00:00                                                                                                                                                               
Saving backup images in ./twrp-backup-2016-10-18--17-55-00/ ...
Saving backup images in ./twrp-backup-2016-10-18--17-55-00/ ...
BusyBox v1.22.1 bionic (2016-08-16 15:18 -0400) multi-call binary.

Usage: mknod [-m MODE] [-Z] NAME TYPE MAJOR MINOR

Create a special file (block, character, or pipe)

	-m MODE	Creation mode (default a=rw)
	-Z	Set security context
TYPE:
	b	Block device
	c or u	Character device
	p	Named pipe (MAJOR and MINOR are ignored)

Saving partition boot (mmcblk0p7), 8 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:01   3.0 MiB/s   5.3 MiB                                                                                                                                         
Traceback (most recent call last):
  File "/usr/local/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.1.0', 'console_scripts', 'tetherback')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python3.4/dist-packages/tetherback/__main__.py", line 3, in <module>
    main()
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 308, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 260, in backup_partition
    devicemd5 = adb.check_output(('shell','cat %s/md5out' % tmpdir)).strip().split()[0]
  File "/usr/local/lib/python3.4/dist-packages/tetherback/adb_wrapper.py", line 30, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "/usr/lib/python3.4/subprocess.py", line 609, in check_output
    output, unused_err = process.communicate(inputdata, timeout=timeout)
  File "/usr/lib/python3.4/subprocess.py", line 947, in communicate
    stdout = _eintr_retry_call(self.stdout.read)
  File "/usr/lib/python3.4/subprocess.py", line 491, in _eintr_retry_call
    return func(*args)
  File "/usr/lib/python3.4/codecs.py", line 319, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte
`

I went with the last stable relase v0.6

**Is there any hope?**

kerlerm, 2016-10-18:
My TWRP is the official 3.0.2-0 build for maguro. Your error seems to have a different cause. Perhaps it's better to open a separate bug report.

To look into fstab for any dm-0 device use` cat /etc/fstab`.

samesamebut, 2016-10-18:
@kerlerm 
But as far as I know (and the developer is telling: http://forum.xda-developers.com/showpost.php?p=68259509&postcount=2055) there is no support for hardware-backed-decryption in this release? How did you achieve to decrypt the data partition in twrp?
In my case I use a different paraphrase for _unlocking_ and _decryption_... maybe here is the _key_ hidden.

Anyway @dlenski the question is how tetherback can handle partitions which are not the type `mmcblk0*`
Is there anyway to do a kind of a "dd"-backup so that it's important which fs-type/data/etc. is inside?

kerlerm, 2016-10-18:
Decrypting worked together with cm-12.1 but TWRP fails to decrypt since I installed cm-13.0. System asks for PIN when booting and decrypts /data just fine. So something must have changed in Android 6 concerning encryption.
See https://github.com/TeamWin/android_device_samsung_maguro/issues/1
Several other bug reports exist mentioning problems in mounting /data on other devices. Just look for "unable to decrypt" or "mount data"
https://github.com/TeamWin/Team-Win-Recovery-Project/issues

So I'm stuck, too.

dlenski, 2016-10-19:
@samesamebut, I'm not sure what's going on here. Tetherback is failing because it's getting unexpected non-text output from `md5sum`.

Please run tetherback` with `-v -v` for all bug reports. This will cause it to print the exact `adb` commands it's running:

   ```
   $ tetherback -v -v
   tetherback v0.9.1
   ADB: ('adb', '-d', 'version')
   Found ADB version 1.0.32
   Using default transfer method: adb exec-out pipe (--exec-out)
   ADB: ('adb', '-d', 'shell', 'uname -r')
   ```

dlenski, 2016-10-19:
I am not sure how you guys expect unmountable, encrypted partitions to be handled. Presumably the right approach is to treat them as binary blobs, just as if you ran with the `--nandroid` option.

kerlerm, 2016-10-19:
As TWRP asks for the decryption password on startup, I assumed it would backup /data unencrypted. But I never checked...

samesamebut, 2016-10-19:
@dlenski 

> Please run tetherback with -v -v for all bug reports. This will cause it to print the exact adb commands it's running:

Get the same output (tried also as root user):

```
# tetherback -v -v
[sudo] password for to: 
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100% Time: 0:00:00                                                                                                                                                                                                          

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       xloader                    128
mmcblk0p2       sbl                       3584
mmcblk0p3       efs                      20480
mmcblk0p4       param                     8192
mmcblk0p5       misc                      4096
mmcblk0p6       dgs                       4096
mmcblk0p7       boot                      8192
mmcblk0p8       recovery                 12224
mmcblk0p9       radio                    16384
mmcblk0p10      system                  669696  /system        ext4
mmcblk0p11      cache                   442368  /cache         ext4
mmcblk0p12      userdata              14198767
mmcblk0p13      metadata                    64
                Total:                15388271

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  -------------------------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="media*" --exclude="*-cache"

Saving backup images in ./twrp-backup-2016-10-19--08-53-13/ ...
Saving backup images in ./twrp-backup-2016-10-19--08-53-13/ ...
BusyBox v1.22.1 bionic (2016-08-16 15:18 -0400) multi-call binary.

Usage: mknod [-m MODE] [-Z] NAME TYPE MAJOR MINOR

Create a special file (block, character, or pipe)

	-m MODE	Creation mode (default a=rw)
	-Z	Set security context
TYPE:
	b	Block device
	c or u	Character device
	p	Named pipe (MAJOR and MINOR are ignored)

Saving partition boot (mmcblk0p7), 8 MiB uncompressed...
  boot.emmc.win: 100% Time: 0:00:01   3.6 MiB/s   5.3 MiB                                                                                                                                                                                    
Traceback (most recent call last):
  File "/usr/local/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.1.0', 'console_scripts', 'tetherback')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python3.4/dist-packages/tetherback/__main__.py", line 3, in <module>
    main()
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 308, in main
    backup_partition(adb, partmap[standard], bp, args.transport, args.verify)
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 260, in backup_partition
    devicemd5 = adb.check_output(('shell','cat %s/md5out' % tmpdir)).strip().split()[0]
  File "/usr/local/lib/python3.4/dist-packages/tetherback/adb_wrapper.py", line 30, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "/usr/lib/python3.4/subprocess.py", line 609, in check_output
    output, unused_err = process.communicate(inputdata, timeout=timeout)
  File "/usr/lib/python3.4/subprocess.py", line 947, in communicate
    stdout = _eintr_retry_call(self.stdout.read)
  File "/usr/lib/python3.4/subprocess.py", line 491, in _eintr_retry_call
    return func(*args)
  File "/usr/lib/python3.4/codecs.py", line 319, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte
```

samesamebut, 2016-10-19:
I have the thought that our (device: maguro) TWRP release behaves differently than  .

@kerlerm Once for sure: we have the same setup (I'm also running CM-13 with encryption) The change was (as I read the 206-pages thread: forum.xda-developers.com/galaxy-nexus/development/recovery-twrp-2-7-1-0-touch-recovery-t1592689/page209) that from now on (CM-13) the encryption on our device is hardware backed. And this "feature" is not (yet) included in TWRP for our device (which is based on omnirom-trees)

Actually I went down here with the hope be able to get a backup from my device. I'm not able to do it "inside" TWRP, but as this software relays somehow on TWRP I guess there is some more digging...

dlenski, 2016-10-19:
@samesamebut... Are you using a very old build of tetherback? The new ones *always* print the version number on startup.

samesamebut, 2016-10-19:
@dlenski the last stable release v0.6

dlenski, 2016-10-19:
Ah, that version is quite old. Please use master. I guess I should push a new release.

samesamebut, 2016-10-19:
```
#tetherback -v -v
tetherback v0.9.1
ADB: ('adb', '-d', 'version')
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
ADB: ('adb', '-d', 'shell', 'uname -r')
Device reports kernel 3.0.101-gc5ca0d9
ADB: ('adb', '-d', 'shell', 'twrp -v')
Device reports TWRP version 3.0.2-0
ADB: ('adb', '-d', 'shell', 'cat /etc/fstab')
ADB: ('adb', '-d', 'shell', 'cd /sys/block; ls -d mmcblk*')
ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/uevent"')
Reading partition map for mmcblk0 (13 partitions)...
  partition map:   0%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p1/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p1/size')
  partition map:   7%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p2/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p2/size')
  partition map:  15%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p3/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p3/size')
  partition map:  23%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p4/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p4/size')
  partition map:  30%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p5/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p5/size')
  partition map:  38%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p6/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p6/size')
  partition map:  46%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p7/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p7/size')
  partition map:  53%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p8/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p8/size')
  partition map:  61%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p9/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p9/size')
  partition map:  69%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p10/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p10/size')
  partition map:  76%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p11/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p11/size')
  partition map:  84%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p12/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p12/size')
  partition map:  92%                                                                                                                ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0/mmcblk0p13/uevent"')
ADB: ('adb', '-d', 'shell', 'cat /sys/block/mmcblk0/mmcblk0p13/size')
  partition map: 100%                                                                                                                
ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0boot0/uevent"')
Reading partition map for mmcblk0boot0 (0 partitions)...
  partition map: 100%                                                                                                                
ADB: ('adb', '-d', 'shell', 'cat "/sys/block/mmcblk0boot1/uevent"')
Reading partition map for mmcblk0boot1 (0 partitions)...
  partition map: 100%                                                                                                                

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       xloader                    128
mmcblk0p2       sbl                       3584
mmcblk0p3       efs                      20480
mmcblk0p4       param                     8192
mmcblk0p5       misc                      4096
mmcblk0p6       dgs                       4096
mmcblk0p7       boot                      8192
mmcblk0p8       recovery                 12224
mmcblk0p9       radio                    16384
mmcblk0p10      system                  669696  /system        ext4
mmcblk0p11      cache                   442368  /cache         ext4
[... cut ...]
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 338, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/usr/local/lib/python3.4/dist-packages/tetherback/tetherback.py", line 229, in backup_partition
    raise RuntimeError("%s: don't know how to mount this partition" % pi.devname)
RuntimeError: mmcblk0p12: don't know how to mount this partition

```

samesamebut, 2016-10-19:
Just out of coriositx I tried a "native" NANDroid Backup

```
# tetherback -N
tetherback v0.9.1
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100%                                                                                                                                            
Reading partition map for mmcblk0boot0 (0 partitions)...
  partition map: 100%                                                                                                                                            
Reading partition map for mmcblk0boot1 (0 partitions)...
  partition map: 100%                                                                                                                                            
Saving backup images in ./nandroid-backup-2016-10-19--18-53-58/ ...
Saving partition boot (mmcblk0p7), 8 MiB uncompressed...
  boot.emmc.gz: 100%   4.7 MiB/s   5.3 MiB                                                                                                                       
Saving partition system (mmcblk0p10), 654 MiB uncompressed...
  system.emmc.gz: 100%   2.6 MiB/s 275.9 MiB                                                                                                                     
Saving partition userdata (mmcblk0p12), 13865 MiB uncompressed...
  userdata.emmc.gz:  38%   4.5 MiB/s   5.2 GB  [##########................................]                                                                                                                   ```

**Still running but far further then any other attempt yet!**

Dump question: How do I roll back a NANDroid Backup? tehterrestore is not ready yet - right?

samesamebut, 2016-10-19:
It worked... somehow...

```
tetherback -N
tetherback v0.9.1
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.0.101-gc5ca0d9
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (13 partitions)...
  partition map: 100%                                                                                                                                            
Reading partition map for mmcblk0boot0 (0 partitions)...
  partition map: 100%                                                                                                                                            
Reading partition map for mmcblk0boot1 (0 partitions)...
  partition map: 100%                                                                                                                                            
Saving backup images in ./nandroid-backup-2016-10-19--18-53-58/ ...
Saving partition boot (mmcblk0p7), 8 MiB uncompressed...
  boot.emmc.gz: 100%   4.7 MiB/s   5.3 MiB                                                                                                                       
Saving partition system (mmcblk0p10), 654 MiB uncompressed...
  system.emmc.gz: 100%   2.6 MiB/s 275.9 MiB                                                                                                                     
Saving partition userdata (mmcblk0p12), 13865 MiB uncompressed...
  userdata.emmc.gz: 100%   3.9 MiB/s   8.9 GiB                                                                                                                   
Backup complete.
```

But the userdata.emmc.gz is not extractable and the file inside the archive is just reported as 1.7GB... this all can't fit

dlenski, 2016-10-23:
@samesamebut:

> How do I roll back a NANDroid Backup? tehterrestore is not ready yet - right?

Right. For now you'll have to "manually" restore the backup, via something like this:

    zcat system.emmc.gz | adb exec-in dd of=/dev/your_system_partition

I've done it several times and it works fine. The 

> But the userdata.emmc.gz is not extractable (boot & system works!) and the file inside the userdata archive is just reported as 1.7GB... this all can't fit

I don't understand how that's possible, because the md5sum is computed on _both_ the device and host side, and it throws an error if there's a mismatch.

Does `md5sum --check userdata.emmc.gz.md5` show that it's valid!??

samesamebut, 2016-10-27:
Here's some update on this topic.
Like the Threadopener I can confirm the backup doesn't work with the official CM-13 (Android 6.0.1).

Now I can also confirm it doesn't work with Android 7.1 [http://forum.xda-developers.com/galaxy-nexus/development/rom-omap4-aosp-project-t3334574] on this device with the same setup (Same TWRP & encrypted data)

The most important point is that also TWRP it self fails to create a backup... and as of my knowledge of this software it relays on twrp... if it doesn't work tetherback can't work... right?

dlenski, 2016-10-27:
> The most important point is that also TWRP it self fails to create a backup... and as of my knowledge of this software it relays on twrp... if it doesn't work tetherback can't work... right?

This isn't very relevant. What TWRP *can't do* is `mount` the encrypted partition as a filesystem. TWRP should *have no problem* accessing its (encrypted) contents as a raw binary blob.

Did you try my suggestion above about checking whether the (encrypted) `userdata.emmc.gz` has a valid `md5sum`?

I need to know the answer to that to understand how to pursue this further.

samesamebut, 2016-10-28:
`Did you try my suggestion above about checking whether the (encrypted) userdata.emmc.gz has a valid md5sum?`
Yes. It has a valid md5sum. The archive itself also has a proper size (~9GB) but the file inside (userdata.emmc) is just reported with a size of 1.7GB and not extractable (breaks with an error while trying to extracted).

dlenski, 2016-10-28:
@samesamebut, ohhhh. Thanks for clarifying that. Hmmm. Some (old) versions of gzip fail to extract files larger than 4 GB.

What happens when you check it with `gunzip -t userdata.emmc.gz`?

dlenski, 2016-10-28:
Are you using MacOS, by the way? The default versions of pretty much all standard Unix shell utilities are broken/ancient on MacOS. :(

samesamebut, 2016-11-02:
>@samesamebut, ohhhh. Thanks for clarifying that. Hmmm. Some (old) versions of gzip fail to extract files larger than 4 GB.

>What happens when you check it with gunzip -t userdata.emmc.gz?

`                                 ` (really! sucks a lots of cpu and then there is no output at all!)

@dlenski using a linux host (*buntu 16.04 based)

dlenski, 2016-11-03:
@samesamebut, thanks for checking. What about "can the device itself gunzip its creation"?

Unfortunately, I'm all out of ideas about this one. I have no idea what would cause your device to create a `.gz` file that doesn't have the right format—but which did not produce an error during the gzip'ing.

closed 2016-12-29 by dlenski