dlenski/tetherback #29: picassowifi: backup fails because  of "unknown option"

kerlerm, 2016-08-21:
Running tetherback with picassowifi throws the following error:

```
$ python3 -m tetherback -R -M -v -v -o /home/nas/.../picasssowifi/
__main__.py v0.7.dev20+ge5a26a4_dirty
ADB: ('adb', '-d', 'version')
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
ADB: ('adb', '-d', 'shell', 'uname -r')
Device reports kernel 3.4.39-g400acec-dirty
ADB: ('adb', '-d', 'shell', 'twrp -v')
Device reports TWRP version 3.0.2-0
ADB: ('adb', '-d', 'shell', 'cat /etc/fstab')
ADB: ('adb', '-d', 'shell', 'cd /sys/block; ls -1d mmcblk*')
ADB: ('adb', '-d', 'shell', 'cat "/sys/block/ls: Unknown option \'-1\'. Aborting./uevent"')
WARNING: don't understand this line from "/sys/block/ls: Unknown option '-1'. Aborting./uevent": "cat: can't open '/sys/block/ls: Unknown option '-1'. Aborting./uevent': No such file or directory"
Reading partition map for ls: Unknown option '-1'. Aborting. (0 partitions)...
  partition map: 100%                                                                                                                     
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/src/git_repo/tetherback/tetherback/__main__.py", line 4, in <module>
    main()
  File "/usr/local/src/git_repo/tetherback/tetherback/tetherback.py", line 305, in main
    plan = plan_backup(args, partmap)
  File "/usr/local/src/git_repo/tetherback/tetherback/tetherback.py", line 178, in plan_backup
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
  File "/usr/local/src/git_repo/tetherback/tetherback/tetherback.py", line 178, in <genexpr>
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
KeyError: 'system'
```

"-1" seems to be an unknown option, but why does this work with other models?

dlenski, 2016-08-21:
Argh. That's frustrating. Yeah, the `-1` option is supported on every other busybox `ls` I've seen. Might be worth filing a bug report with TWRP/omnirom.

I'll have to try to come up with another lowest-common-denominator option for getting all the `mmcblk*` devices.

Any suggestions? :(

kerlerm, 2016-08-22:
Would that be the right place to file the bug? 
https://github.com/TeamWin/android_device_samsung_picassowifi/issues

dlenski, 2016-08-22:
I believe so, yes.

I honestly don't know much about the TWRP development process, but presumably they configured `busybox` with slightly different capabilities on the picassowifi build.

Before filling the bug, you should definitely confirm that `busybox help ls` doesn't show the `-1` option on picassowifi, but does show it for the same TWRP release on other devices.

kerlerm, 2016-09-04:
bug filed:
https://github.com/TeamWin/android_device_samsung_picassowifi/issues/1

closed 2016-09-10 by dlenski