dlenski/tetherback #30: Issue on Archlinux and Samsung Note 4

hverjus, 2016-08-26:
I desesperate to find such a tool/script. I'am on Archlinux and have this error interruption the tetherback(up):

```
$ tetherback
tetherback v0.8
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.40-
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (27 partitions)...
  partition map: 100%                                                         
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                         
Reading partition map for mmcblk1 (1 partitions)...
  partition map:   0%                                                         Traceback (most recent call last):
  File "/usr/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.8', 'console_scripts', 'tetherback')()
  File "/usr/lib/python3.5/site-packages/tetherback/tetherback.py", line 302, in main
    partmap = build_partmap(adb)
  File "/usr/lib/python3.5/site-packages/tetherback/tetherback.py", line 144, in build_partmap
    partname = d['PARTNAME'].lower()
KeyError: 'PARTNAME'
$
```
I installed the latest development release with python3 (pip3) as explained in the installation  notes.

dlenski, 2016-08-26:
The kernel name for your device appears to be cut off (`3.10.40-`). What is your device?

The error suggests that the mmcblk uevent files on your device are not formatted in the usual way. I can make a workaround to just ignore this partition, but I would like to know if any other expected fields are missing.

Could you please run this and show me the output?
`$ adb shell cat /sys/block/mmcblk1/mccblk1p1/uevent`

dlenski, 2016-08-26:
You can use this branch ([feature/skip_partitions_with_incomplete_uevent](../tree/feature/skip_partitions_with_incomplete_uevent)) for a temporary workaround that will just ignore this one partition with an incomplete description.

hverjus, 2016-08-26:
My device is a Samsung Galaxy note 4. I did:

`$ adb shell cat /sys/block/mmcblk1/mccblk1p1/uevent
`
That shows:
```
MAJOR=179
MINOR=64
DEVNAME=mmcblk1
DEVTYPE=disk
NPARTS=1
```
Thank you a lot :-) I will try the workaround and will let you know.

hverjus, 2016-08-26:
Great ! The workaround did the trick and generates the archive files (large file for /data) of the partitions. Putting the device in recovery mode again, how can we process the restore (from a PC terminal directly to the device)? I thought to put all the archive files on a sdcard and process a restore from TWRP but sdcard doesn't accept files larger to 4 Go (vfat filesystem) and it would be nice to avoid sdcard fro restoring the device as the backup process does.

dlenski, 2016-08-26:
Glad to hear it!

I don't have a complete restore script yet. I've started working on this and you can see some details over at issue #5, but no guarantees as to when it'll be ready.

I have done a "manual" restore myself by piping the data files into the device with `adb`, and basically doing the reverse of what tetherback already does itself. It works as intended.

hverjus, 2016-08-26:
If you have some insights (command lines...) of what you do I'm interested in :-) Thank's

closed 2016-09-10 by dlenski