dlenski/tetherback #31: Stardard partitions not found

SkorpioDav, 2016-09-06:
Nvidia Tablet K1

Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.33-gff23870
Device reports TWRP version 3.0.2-2
Reading partition map for mmcblk0 (24 partitions)...
  partition map: 100% Time: 0:00:10

Partition map:

BLOCK DEVICE    PARTITION NAME             SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  -----------------------  ------------  -------------  --------
mmcblk0p1       tos                              6144
mmcblk0p2       eks                              2048
mmcblk0p3       fb                               2048
mmcblk0p4       nct                              2048
mmcblk0p5       wb0                               512
mmcblk0p6       rp1                              4096
mmcblk0p7       rp2                              4096
mmcblk0p8       rp3                              8192
mmcblk0p9       rp4                              8192
mmcblk0p10      sos                             16384
mmcblk0p11      dtb                              4096
mmcblk0p12      lnx                             16384
mmcblk0p13      app (standard system)         2097152  /system        ext4
mmcblk0p14      cac (standard cache)           262144  /cache         ext4
mmcblk0p15      msc                              2048
mmcblk0p16      usp                             32768
mmcblk0p17      mda                              2048
mmcblk0p18      fct                             16384
mmcblk0p19      lbp                              4096
mmcblk0p20      chg                              4096
mmcblk0p21      fbp                              4096
mmcblk0p22      fcg                              4096
mmcblk0p23      uda (standard userdata)      12864000  /data          ext4
mmcblk0p24      gpt                               512
                Total:                       15367680

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  -------------------------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="media*" --exclude="*-cache"

usage: tetherback-script.py [-h] [-s DEVICE_ID] [-o OUTPUT_PATH] [-N] [-0]
                            [-V] [-v] [-f] [-t | -x | -6 | -P] [-M] [-D] [-R]
                            [-C] [-U] [-S] [-B] [-X NAME]
tetherback-script.py: error: Standard partitions were requested for backup, but not found in the partition map: bootPlease report this issue at https://github.com
/dlenski/tetherback/issues
Please post the entire output from tetherback!

dlenski, 2016-09-06:
Thank you very much for reporting this! The partitions on your device are named in an unusual way, which is why the `boot` partition is not auto-detected. I may need to add further tricks to try to autodetect the boot partition, which contains an Android/Linux kernel boot image.

I suspect that `mmcblk0p12`/`lnx` is your boot partition, but I'm not totally sure.

Could you please run this adb command (while booted in TWRP recovery)?

`$ adb exec-out "dd if=/dev/block/mmcblk0p12 bs=65536 count=1"|file -`

If this is your boot partition, then `file` should describe it as something like "Linux/Android kernel image".

If this is not your boot partition, then please try this on some of the other partitions and tell me which one you find to be the boot partition.

SkorpioDav, 2016-09-07:
Thank you for the answer and thank you very much for your program. Sorry for my previous essential post but I was in hurry. I have tried the command you provided but it gives me an error:

 "file" is not recognized as an internal or external command, operable program or batch file. 

I have tried to run the command without the "|file -" at the end to see if it was just a wrong parameter but the computer started to bleep until I closed the prompt. Did I make any damage to my tablet or PC?

dlenski, 2016-09-07:
What OS are you running, on your laptop? `file ` is a utility to identify the type of a file based on its header/contents, which should be available on any Linux/Unix/MacOS system, but not by default under windows.

dlenski, 2016-09-07:
If you don't have `file` or something similar, this is an alternative:

```sh
adb exec-out "dd if=/dev/block/mmcblk0p12 bs=65536 count=1" > first_256_bytes_of_partition.bin
```

… then email me that file, or just attach it to the issue, and I can figure it out.

(Hopefully my guess is right and this is really just the generic `boot` partition, but do keep in mind that there's a *small* chance that I'm wrong and that this block of data will contain some device-specific information like your tablet's serial number or IMEI, etc.)

SkorpioDav, 2016-09-07:
I'm running Windows 7, I found this program for Win that should do the same as File http://mark0.net/soft-trid-e.html
This is the output for the first command (I saved it into a file and then analyzed it with "Trid")

adb exec-out "dd if=/dev/block/mmcblk0p12 bs=65536 count=1" > file.bin

TrID/32 - File Identifier v2.24 - (C) 2003-16 By M.Pontello
Definitions found:  7074
Analyzing...

Collecting data from file: file.bin
100.0% (.IMG) Android boot image (8000/1)

Could you please check if the program does what we need? (It is available for linux too if you want to compare it to "File"). Meanwhile I'll try to install ADB to my old eeepc with linux and run the command from that.

dollarusername, 2016-09-07:
I have a similar issue, with a Pixel C running RemixOS.  Note I have NOT replaced the stock bootloader, but rather booted into TWRP via "fastboot boot twrp-3.0.0-0-dragon-ryu-02112016-1.img".  I had bootloop issues replacing TWRP earlier so reverted to stock.

tetherback -N
tetherback v0.8
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.18.0-00002-gacae30e
Device reports TWRP version 3.0.0-0
Reading partition map for mmcblk0 (11 partitions)...
  partition map: 100%
Reading partition map for mmcblk0boot0 (0 partitions)...
  partition map: 100%
Reading partition map for mmcblk0boot1 (0 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%

Partition map:

BLOCK DEVICE    PARTITION NAME             SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  -----------------------  ------------  -------------  --------
mmcblk0p1       kern-a                          32768
mmcblk0p2       kern-b                          32768
mmcblk0p3       recovery                        32768
mmcblk0p4       app (standard system)         3670016  /system        ext4
mmcblk0p5       vnr                            311296  /vendor        ext4
mmcblk0p6       cac (standard cache)           409600  /cache         ext4
mmcblk0p7       uda (standard userdata)      25934848  /data          ext4
mmcblk0p8       md1                             65536
mmcblk0p9       lnx                             32768
mmcblk0p10      msc                              4096
mmcblk0p11      pst                               512
                Total:                       30526976

Backup plan:

PARTITION NAME    FILENAME          FORMAT
----------------  ----------------  -----------------
boot              boot.emmc.gz      gzipped raw image
system            system.emmc.gz    gzipped raw image
userdata          userdata.emmc.gz  gzipped raw image

usage: tetherback [-h] [--version] [-s DEVICE_ID] [-o OUTPUT_PATH] [-N] [-0]
                  [-V] [-v] [-f] [-t | -x | -6 | -P] [-M] [-D] [-R] [-C] [-U]
                  [-S] [-B] [-X NAME] [--extra-raw NAME]
tetherback: error: Standard partitions were requested for backup, but not found in the partition map: bootPlease report this issue at https://github.com/dlenski/tetherback/issues
Please post the entire output from tetherback!

dlenski, 2016-09-07:
Thanks to both of you.

Looks like both of your devices use `lnx` as the name of the partition normally called `boot`, containing the Linux kernel. I'll cook up some way to autodetect this as the boot partition.

(@dollarusername, can you please confirm as well?)

```sh
$ adb exec-out "dd if=/dev/block/mmcblk0p12 bs=65536 count=1" > first_256_bytes_of_partition.bin
$ file first_256_bytes_of_partition.bin     # or whatever equivalent tool you have on your OS
```

dollarusername, 2016-09-08:
Hmm, well using dd didn't work that way because there wasn't root privs to read the block device:
$adb exec-out id
uid=2000(shell) gid=2000(shell) groups=2000(shell),1004(input),1007(log),1011(adb),1015(sdcard_rw),1028(sdcard_r),3001(net_bt_admin),3002(net_bt),3003(inet),3006(net_bw_stats) context=u:r:shell:s0

..anyway, I got a root shell on the device and ran:
dd if=/dev/block/mmcblk0p9 bs=65536 count=1 > /data/x

After pulling that file back running file on it only says it's data:
x: data

Strings comes back with absolutely nothing as well.

dlenski, 2016-09-08:
@dollarusername, you should always have root privileges via `adb` if booted into TWRP… ?

Can you tell which of the other partitions *is* the Android boot image? `mmcblk0p1` or `mmcblk0p2`, maybe? The image should start with this 8-byte magic, `ANDROID!`.

dollarusername, 2016-09-08:
duh.  Yeah, I wasn't booted into TWRP.  Anyway, it seems for my Pixel (running Remix OS, based on 6.0.1).  I assume it's CHROMEOS that you want to see for this device:

$ for var in 1 2 3 4 5 6 7 8 9 10 11; do echo Doing mmcblk0p$var;  adb exec-out "dd if=/dev/block/mmcblk0p$var bs=65536 count=1" |strings |head -2; done
Doing mmcblk0p1
CHROMEOS
iHiA*R
Doing mmcblk0p2
1+0 records in
1+0 records out
Doing mmcblk0p3
CHROMEOS
iHiA*R
Doing mmcblk0p4
vTozsystem
/system
Doing mmcblk0p5
vendor
1+0 records in
Doing mmcblk0p6
/cache
1+0 records in
Doing mmcblk0p7
_MDOP
7_l`
Doing mmcblk0p8
aes-cbc-essiv:sha256
(s=
Doing mmcblk0p9
1+0 records in
1+0 records out
Doing mmcblk0p10
1+0 records in
1+0 records out
Doing mmcblk0p11
^gXqMr
hd\Ft

dlenski, 2016-09-08:
Aha, RemixOS… I know very little about it (didn't even realize it was related to ChromeOS, actually).

Does `mmcblk0p1` actually contain the **whole** kernel? The names `kern-a` and `kern-b` suggest that it might actually be split across two partitions, for some reason.

dollarusername, 2016-09-09:
Oddly it's not related to ChromeOS at all as far as I know-- it's basically straight Android 6.0.1 but they've added a window manager and a few other things.  The PixelC apparently was designed to run ChromeOS although that has never been an actual option since it was released AFAIK.

I put both p1 and p2 up on dropbox (grabbed via:
adb exec-out "dd if=/dev/block/mmcblk0p? bs=1024 count=32" > mmcblk0p?.dd

$ md5sum *
77dc30f622d6f7d63cc27b535edfe6fc  mmcblk0p1.dd
b16a4cbd98dd36ce1183698d7dfabee8  mmcblk0p2.dd

https://www.dropbox.com/s/e7f4xsheugfniy4/mmcblk0p1.dd?dl=0
https://www.dropbox.com/s/cqtyx6sa449j0ab/mmcblk0p2.dd?dl=0

hope that helps!

dlenski, 2016-09-09:
Hmmm... I don't recognize the formats of those file. And I really don't think I understand this device well enough to make tetherback just "do the right thing automatically." I don't think that it is using the [standard-ish Android boot process](http://newandroidbook.com/Articles/aboot.html), but probably something that's sort of a hybrid of the ChromeOS and Android boot processes.

Until I can figure out a way to auto-identify the boot partition, @dollarusername, you're going to have to use `tetherback --no-boot -X kern-a -X kern-b` to include these partitions and avoid the error about missing `boot` partitions.

closed 2016-10-23 by dlenski

SkorpioDav, 2016-10-23:
Hello again, I have tried the latest version but I have still the problem, did you fix it? Is there a command line I can use to bypass the problem and backup my Shield K1 with your software? Thank you.

dlenski, 2016-10-24:
@SkorpioDav, it *should* work for you with the following command line. This tells tetherback not to look for a standard-named `boot` partition, but to backup the `lnx` partition instead…

```sh
$ tetherback --no-boot -X lnx
```

I'm reluctant to start adding exceptions for model-specific partitions that don't have standard names, because I don't have a good framework for testing or maintaining them, or validating which specific models they work for.

aperomsik, 2016-11-21:
Just as an additional data point, 2012 Nexus 7 also has "lnx" and for that partition...

```
$ adb exec-out "dd if=/dev/block/mmcblk0p2 bs=65536 count=1" | file -
/dev/stdin: Android bootimg, kernel (0x10008000), ramdisk (0x11000000), page size: 2048
```

dlenski, 2016-11-21:
Thanks, @aperomsik, for another useful data point. A pull request for dealing with alternative partition names in a clean and extensible way would be welcome :)