dlenski/tetherback #32: Wireless ADB

No3x, 2016-09-12:
I try to make backups via wireless adb connection.
Everything seems ok with the connection to the device:
```
adb devices
List of devices attached
192.168.178.34:5555     device
```
The output of this also fine:
```
adb shell uname -r
3.10.84-perf+
````
But this doesn't work:
```
tetherback.exe -M -R
tetherback-script.py v0.9
Found ADB version 1.0.31
Using default transfer method: adb TCP forwarding (--tcp)
error: device not found
usage: tetherback-script.py [-h] [--version] [-s DEVICE_ID] [-o OUTPUT_PATH]
                            [-N] [-0] [-V] [-v] [-f] [-t | -x | -6 | -P] [-M]
                            [-D] [-R] [-C] [-U] [-S] [-B] [-X NAME]
                            [--extra-raw NAME]
tetherback-script.py: error: Device not connected. adb returns an error:

        adb -d shell
```
Is there something I can try? Should I just try with linux?

dlenski, 2016-09-12:
Check the README. **You must boot into TWRP recovery in order to use tetherback.** TWRP recovery only supports a USB connection, *not* wireless ADB.

That is why tetherback invokes adb with the `-d` option, because the `-d` option tells adb only to use a USB connection:

```
Android Debug Bridge version 1.0.32

 -a                            - directs adb to listen on all interfaces for a connection
 -d                            - directs command to the only connected USB device
                                 returns an error if more than one USB device is present.
```

No3x, 2016-09-12:
Yes I checked the readme. But especially the -t param made me think it must work not only with usb.

Running the backup with usb was fine. But the battery drained so fast I was not able to complete the backup. So I was trying to power the phone while connecting it via wireless adb.

dlenski, 2016-09-13:
Ah, I see. Yes, I should clarify that with the `-t` option it still only works over USB.

How does your device lose its charge while connected via USB? Doesn't it charge while the backup is made?

No3x, 2016-09-13:
It runs really hot (85°C). The transfer rate is ~7,5Mb/s and my media partition is 64GB. The phone was not fully charged.
I cleaned up a bit yesterday and freed a lot of disk. The media partition is 24GB now.

dgw, 2016-09-13:
@dlenski It's been my experience that USB ports tend to provide insufficient power for mid-to-high-end phones while under load. So even though the device is "charging" while connected, it may actually be using more power than it can draw from the data port (limited to 500mA while connected as a data device, unless that has changed for USB 3.0).

No3x, 2016-09-13:
@dgw I can confirm this on a USB 3.0 Port with OnePlus 2

dlenski, 2016-09-13:
Hmm. I've never had a problem charging the Nexus 5 off of a USB port while tethering or backing up or otherwise drawing a fairly high current. Then again, I'm always using Linux, which gives power to USB ports more generously than Windows.

In any case, there isn't anything I can do to fix this; since TWRP only supports adb-over-USB, tetherback can only support adb-over-USB.

commit a185a4b1 (2016-09-14): clarify that USB connection is required in error message (closes #32)
closed 2016-09-14 by dlenski