dlenski/tetherback #33: tetherback failed, please help... Galaxy S5

pwndguy, 2016-09-14:
When I ran tetherback, this error showed up...

`tetherback-script.py v0.9
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-twrp-klte-unified-1.7
Device reports TWRP version 3.0.2-2
Reading partition map for mmcblk0 (26 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%
Reading partition map for mmcblk1 (1 partitions)...
  partition map:   0%                                                          Traceback (most recent call last):
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\Scripts\tetherback-script.py", line 9, in <module>
    load_entry_point('tetherback==0.9', 'console_scripts', 'tetherback')()
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\lib\site-packages\tetherback\tetherback.py", line 316, in main
    partmap = build_partmap(adb)
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\lib\site-packages\tetherback\tetherback.py", line 152, in build_partmap
    print("WARNING: partition %sp%d has no PARTNAME in its uevent file" % (mmcblk, ii, please_report), file=stderr)
TypeError: not all arguments converted during string formatting`

dlenski, 2016-09-14:
That's my bad. Please re-pull with the tiny fix I just committed.

When you re-run it, it'll give you a warning about a partition with an unknown name (on `mmcblk1`, which means it's probably not a partition relevant for backup) but should otherwise work fine.

pwndguy, 2016-09-14:
Wow, that was really fast, and it seems to have worked, I should only have 1 file in the folder it created right?

Here's the new log just in case...

`tetherback-script.py v0.9.1
Found ADB version 1.0.32
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-twrp-klte-unified-1.7
Device reports TWRP version 3.0.2-2
Reading partition map for mmcblk0 (26 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%
Reading partition map for mmcblk1 (1 partitions)...
  partition map:   0%                                                          WARNING: partition mmcblk1p1 has no PARTNAME in its uevent file
Please report this issue at https://github.com/dlenski/tetherback/issues
Please post the entire output from tetherback!
  partition map: 100%
Saving backup images in .\twrp-backup-2016-09-14--16-41-25/ ...
Saving partition boot (mmcblk0p15), 13 MiB uncompressed...
  boot.emmc.win: 100%   6.3 MiB/s  10.0 MiB
Traceback (most recent call last):
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\Scripts\tetherback-script.py", line 9, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\lib\site-packages\tetherback\tetherback.py", line 338, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "C:\Users\TYLER\AppData\Local\Programs\Python\Python35-32\lib\site-packages\tetherback\tetherback.py", line 292, in backup_partition
    raise RuntimeError("md5sum mismatch (local %s, device %s)" % (localmd5, devicemd5))
RuntimeError: md5sum mismatch (local 11241fc187e7bf5096f8f00dba09e655, device 9491a950e9e6f4568e3b05d5ab7bd967)`

dlenski, 2016-09-15:
No, it has failed, though for a more interesting reason.

The contents of the boot partition have been corrupted during the transfer. (I'm really glad I added checking for this a few months ago :+1:). 

This could be (1) something your device is doing wrong, (2) something adb is doing wrong, or (3) something tetherback is doing wrong.

A problem with `adb` should be the easiest to rule out. Could you try running with the `--base64` (very slow) or `--tcp` transfer methods and see if it succeeds? If so, then there's something wrong with your version of adb.

```
Data transfer methods:
  The default is --exec-out with adb v1.0.32 or newer, and --tcp with older
  versions. If you have problems, please try --base64 for a slow but
  reliable transfer method (and report issues at
  http://github.com/dlenski/tetherback/issues)

  -t, --tcp             ADB TCP forwarding (fast, should work with any host
                        OS, but prone to timing problems)
  -x, --exec-out        ADB exec-out binary pipe (should work with any host
                        OS, but only with newer versions of adb and TWRP)
  -6, --base64          Base64 pipe (very slow, should work with any host OS)
  -P, --pipe            ADB shell binary pipe (fast, but probably only works
                        on Linux hosts)
``

pwndguy, 2016-09-15:
sorry for the delay, I had to do chores for a while... I'm pretty sure it worked this time, it has the 3 files at the correct sizes, with the --tcp btw, but here's the log anyway...

```
tetherback-script.py v0.9.1
Found ADB version 1.0.32
Device reports kernel 3.4.0-twrp-klte-unified-1.7
Device reports TWRP version 3.0.2-2
Reading partition map for mmcblk0 (26 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%
Reading partition map for mmcblk1 (1 partitions)...
  partition map:   0%                                                          WARNING: partition mmcblk1p1 has no PARTNAME in its uevent file
Please report this issue at https://github.com/dlenski/tetherback/issues
Please post the entire output from tetherback!
  partition map: 100%
Saving backup images in .\twrp-backup-2016-09-14--17-45-28/ ...
Saving partition boot (mmcblk0p15), 13 MiB uncompressed...
  boot.emmc.win: 100%   6.6 MiB/s  10.0 MiB
Saving tarball of mmcblk0p23 (mounted at /system), 2700 MiB uncompressed...
  system.ext4.win: 100%   4.7 MiB/s   1.4 GiB
Saving tarball of mmcblk0p26 (mounted at /data), 11622 MiB uncompressed...
  data.ext4.win: 100%   5.0 MiB/s   6.3 GiB
Backup complete.
```

dlenski, 2016-09-15:
Great, thanks! Okay, so it's pretty clear that this is a problem with your version of ADB, and *not* with either tetherback or with your device.

Can you tell me where you got this version of `adb`? And what version of Windows you're on? 

(I haven't seen this with adb 1.0.32, even from other Windows users.)

pwndguy, 2016-09-15:
Windows 10, and i got it from [http://forum.xda-developers.com/showthread.php?t=2588979](url)

dlenski, 2016-09-15:
Thanks, I'll keep an eye out for similarly buggy versions.

closed 2016-09-15 by dlenski