dlenski/tetherback #34: Publish to PyPi

ChillarAnand, 2016-10-24:
Great work. Can you publish to PyPi?

dlenski, 2016-10-24:
What's the advantage of publishing to PyPi? Tetherback isn't exactly a module or otherwise a very reusable piece of Python code, since it's dependent on the external `adb`.

You can install it with `pip3` quite easily even without having it published to PyPi.

ChillarAnand, 2016-10-30:
Not much. My friend told me about this package. I tried to do pip install and it didn't work. I googled it and then got link and installed it. Thought it would be easier for installations.

closed 2016-10-30 by ChillarAnand

dlenski, 2016-10-31:
Did the [pip installation instructions](https://github.com/dlenski/tetherback#requirements-and-installation) not work? What platform?

ChillarAnand, 2016-10-31:
`pip install tetherback` didn't work. `pip install git+https://github.com/dlenski/tetherback/` worked

dgw, 2016-11-01:
@dlenski One argument for uploading releases to PyPI is that it greatly simplifies upgrading. For packages not listed on PyPI, one must `pip install git+[whatever_url]` every time an update is required. Packages that _are_ listed on PyPI can be `pip install -U packagename`'d without having to remember (or copy) a URL every time—simply knowing the package name is sufficient.

I don't know if that's sufficient reason to consider uploading tetherback or not, but I know that my installed versions of packages fetched from outside PyPI tend to be much more outdated because it's such a hassle to make pip upgrade them. (pip lacking a proper "upgrade everything" command is another issue, literally: pypa/pip#59)

dlenski, 2016-11-01:
@dgw, good point. Are there other examples of things that are more like standalone tools, rather than libraries, which are on PyPi?