dlenski/tetherback #35: Be more explicit about excluding /data/media*

Blender3D, 2016-10-28:
I lost a bunch of data recently because TWRP excludes `/data/media*` from its backups. The developers of TWRP [don't see this as a serious issue](https://github.com/TeamWin/Team-Win-Recovery-Project/issues/276), but excluding the one thing that I actually care about from my backups is dangerous default behavior.

How do you feel about renaming `--media` to `--include-media` and making the user pick either `--include-media` or `--exclude-media` when doing a TWRP-compatible backup?

dlenski, 2016-10-29:
Ah, I hadn't thought about that. You're right that it's not a safe default behavior, especially for `tetherback` which isn't limited by storage space on the device itself.

It's probably best to include `/data/media*` by default and add a `--no-media` option (similar naming to `--no-boot`, etc.).

Want to take a crack at a PR to fix that?

dgw, 2016-10-29:
_silently forks and clones the project to have a little fun making a PR_

I've been meaning to practice with argparse anyway 😀

PR #36 (2016-10-29): Include /data/media* by default
Removes `-M` / `--media` flags and replaces them with `-E` / `--no-media` flags.

`-E` chosen instead of `-M` to avoid confusion and issues with scripts, since the new flag's behavior is the exact opposite of the old.

Intended to resolve #35.
closed 2016-10-29 by dlenski