dlenski/tetherback #37: Adding requirements.txt

dgw, 2016-11-02:
If a package contains a `requirements.txt` file, pip can automatically install python dependencies. With reference to #34, it would make the installation process slightly easier, even if tetherback itself is not in PyPI.

The stated requirements are progressbar2 and tabulate, but unknown which versions of each. @dlenski I'm happy to create and test a requirements file tonight if you have any hints what minimum (and/or maximum) versions of those dependencies work.

dlenski, 2016-11-03:
My understanding of Python packaging is way out of date and spotty, but I thought [that the `install_requires` bit](https://github.com/dlenski/tetherback/blob/master/setup.py#L42) was supposed to specify this:

```python
      install_requires=[ 'progressbar2>=3.6', 'tabulate' ],
```

I could've sworn there was an option to `pip` to install the dependencies as well, but can't find it at the moment.

dgw, 2016-11-03:
I'm not even sure that the documentation is correct. I just read through a few docs on the file vs. setup parameter, and they indicate that the text file isn't used unless pip is given the `-r` argument. Yet, I `pip install` packages with only `requirements.txt` all the time, without that argument, and the dependencies are installed.

I might have to look into it more. It's true that I usually get Python 2.7 on servers, so even though pip should be consistent across versions it's possible that the behavior actually differs between py2 and py3.

dlenski, 2016-11-03:
Yeah, I found the documentation frustratingly vague and inconsistent. Does your `pip3` *not* seem to use the `install_requires` argument?

If you want to submit a PR with the correct `requirements.txt`, I'm all for it.

`progressbar2>=3.6` (cf. #12 where we figured out that this was the minimum version, as well as #20 and [python-progressbar issue #76](https://github.com/WoLpH/python-progressbar/issues/76) where there there was a small bug in `progressbar2`)
`tabulate` (not sure what version is required, because I haven't got any reports of one that's so old that it doesn't :+1:)

dlenski, 2016-11-04:
Thanks @dgw. I'm with you on not duplicating the requirements unless necessary.

Do you have any suggestions for how to word the requirements?

dgw, 2016-11-04:
@dlenski I went ahead and took a rewrite pass in #38. End of the night for me, but I look forward to seeing your thoughts there tomorrow.

Closing this issue as invalid/unnecessary.

closed 2016-11-04 by dgw