dlenski/tetherback #39: Device with non mcblk partition table (Galaxy S6)

Xyv, 2016-11-07:
Hello,

First: Thanks for the script! 

I have a Galazy S6 in my possession and they named the partition list as 'sda#' not 'mcblk0p#.
Listing partitions in adb shell returns:
```bash
~ # ls -l /dev/block/platform/15570000.ufs/by-name/

lrwxrwxrwx    1 root         root                15 Nov  7 11:03 BOOT -> /dev/block/sda5
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 BOTA0 -> /dev/block/sda1
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 BOTA1 -> /dev/block/sda2
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 CACHE -> /dev/block/sda16
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 DNT -> /dev/block/sda10
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 EFS -> /dev/block/sda3
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 HIDDEN -> /dev/block/sda17
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 OTA -> /dev/block/sda7
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 PARAM -> /dev/block/sda4
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 PERSDATA -> /dev/block/sda13
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 PERSISTENT -> /dev/block/sda11
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 RADIO -> /dev/block/sda8
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 RECOVERY -> /dev/block/sda6
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 SBFS -> /dev/block/sda14
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 STEADY -> /dev/block/sda12
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 SYSTEM -> /dev/block/sda15
lrwxrwxrwx    1 root         root                15 Nov  7 11:03 TOMBSTONES -> /dev/block/sda9
lrwxrwxrwx    1 root         root                16 Nov  7 11:03 USERDATA -> /dev/block/sda18
```

```bash
~ # cat /proc/partitions
major minor  #blocks  name

   8        0   31240192 sda
   8        1       4096 sda1
   8        2       4096 sda2
   8        3      20480 sda3
   8        4       8192 sda4
   8        5      28672 sda5
   8        6      34816 sda6
   8        7       8192 sda7
   8        8      43008 sda8
   8        9       1024 sda9
   8       10       1024 sda10
   8       11        768 sda11
   8       12        256 sda12
   8       13       9216 sda13
   8       14      15360 sda14
   8       15    3788800 sda15
 259        0     204800 sda16
 259        1      10240 sda17
 259        2   27045888 sda18
   8       32       4096 sdc
   8       16       4096 sdb
```

So a few adjustments to your script were needed in `build_partmap`
 ```python
mmcblks = adb.check_output(('shell','cd /sys/block; ls -d mmcblk*')).split()
```
to
```python
mmcblks = adb.check_output(('shell','cd /sys/block; ls -d sda*')).split()
```
and removing the 'p' in the calls to specific partitions : '%s**p**%d' % (mmcblk, partn) to '%s%d' % (mmcblk, partn)

running `tetherback -0` works fine afterwards:
```bash
$ python3 -m tetherback -0
__main__.py v0.9.1
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.101-arter97-13.1-04960-ge491d042
Device reports TWRP version 3.0.2-0
Reading partition map for sda (18 partitions)...
  partition map: 100%

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
sda1            bota0                     4096
sda2            bota1                     4096
sda3            efs                      20480
sda4            param                     8192
sda5            boot                     28672
sda6            recovery                 34816
sda7            ota                       8192
sda8            radio                    43008
sda9            tombstones                1024
sda10           dnt                       1024
sda11           persistent                 768
sda12           steady                     256
sda13           persdata                  9216
sda14           sbfs                     15360
sda15           system                 3788800  /system        ext4
sda16           cache                   204800  /cache         ext4
sda17           hidden                   10240
sda18           userdata              27045888  /data          ext4
                Total:                31228928

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  ------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="*-cache"
```

Also I don't know if it is intended but in my case `really_umount` in `backup_partition` is called for every partition whether it is mounted or not resulting in:
`subprocess.CalledProcessError: Command '('adb', '-d', 'shell', 'umount /dev/block/sda5 2>/dev/null && echo ok')' returned non-zero exit status 1`
As (I) expected, running the command in adb shell returns nothing if partition not mounted and 'ok' otherwise.

Dirty trick is to comment the `really_umount` part.

Also got same problem as issue #28 when doing a TWRP backup, no problem with nandroid.

Don't know if you plan on updating your script for this particular case but I thought others could be interested.

dlenski, 2016-11-13:
Thanks, @Xyv, this is a very useful and detailed report. I actually didn't know that there were any Android devices using `/dev/sda*` instead of `/dev/block/mmcblk*`.

When I originally wrote this, I went back and forth on whether to use `cat /proc/partitions` or `ls /sys/block` as the starting point for the partition list. It seems like this may need to be revised to support more devices.

> Also I don't know if it is intended but in my case really_umount in backup_partition is called for every partition whether it is mounted …

That *is* intentional, actually. Attempting to `umount` an unmounted partition should be harmless. I am surprised that `adb` is returning an error. What version are you using? (`adb version`)

> Also got same problem as issue #28 when doing a TWRP backup, no problem with nandroid.

Issue #28 got rather long and probably should be split up. Which part of the problem did you run into with a TWRP backup? Was it the issue with being unable to mount the encrypted partition, or something else?

Xyv, 2016-11-14:
Yes I was also surprised about this setup of the device but well...

> That is intentional, actually. Attempting to umount an unmounted partition should be harmless. I am surprised that adb is returning an error. What version are you using? (adb version)

I agree it should be harmless but somehow adb is returning empty string in my case. adb should be latest version:
```
$ adb version
Android Debug Bridge version 1.0.36
Revision 84e3321d5db3-android
```

Forgot to mention something quite important here...running on a mac. I often have to patch my softwares intended for linux with mac-specific cases(and sometimes even mac osx version specific too)...lot of fun.
```
 $ uname -a
Darwin xxx.local 15.6.0 Darwin Kernel Version 15.6.0: Mon Aug 29 20:21:34 PDT 2016; root:xnu-3248.60.11~1/RELEASE_X86_64 x86_64
```
> Issue #28 got rather long and probably should be split up. Which part ...

I was referring to the crash mentioned in the OP when trying to mount the /data partition.
I actually didn't check whether the partition is encrypted or not...I'll let you know.

> A patch or pull request to handle both cases elegantly would be very welcome :-D

Will do that when I find some time to play around a bit more (could take some times...;-))

And thanks for following up on this piece of software !

GigabyteProductions, 2017-03-26:
Google Pixel XL also uses sd* block devices.