dlenski/tetherback #41: F2FS backups

DerNoobikeks, 2017-02-07:
May be a stupid question, but am i able to make backups of F2FS partitions with the TWRP backup method?

dlenski, 2017-02-08:
I don't own an Android device that uses [F2FS](https://en.wikipedia.org/wiki/F2FS), so I have no way to test whether or not it works.

The code that extracts the partition contents is relatively filesystem-agnostic, but what probably _will_ need some tweaking is the code that gets the list of partitions from the device. (Android devices are very frustratingly inconsistent about how they list partitions.)

If you can do the following, I can probably figure out what would be needed to support your device:

1. Boot into TWRP recovery
2. Run `adb shell cat /proc/partitions` and `adb shell ls /sys/block/*` and `adb uname -a`
3. Show the output here

dlenski, 2017-02-28:
I just tested with a device that has an F2FS partition for `/data`, and it works fine with the small tweak in 67c7a88.

closed 2017-02-28 by dlenski