dlenski/tetherback #42: RuntimeError: mmcblk0p42: don't know how to mount this partition

Cubox-, 2017-02-12:
```
<cubox@Sleipnir>-<~> λ tetherback -v
tetherback v0.9.1
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.40-gee46fae
Device reports TWRP version 3.0.3-0
Reading partition map for mmcblk0 (42 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       modem                   114688  /firmware      ext4
mmcblk0p2       metadata                 16384
mmcblk0p3       sbl1                       384
mmcblk0p4       sdi                         56
mmcblk0p5       sec                         16
mmcblk0p6       ddr                         32
mmcblk0p7       aboot                     1024
mmcblk0p8       rpm                        256
mmcblk0p9       utags                      512
mmcblk0p10      tz                         500
mmcblk0p11      pada                      4156
mmcblk0p12      sbl1bak                    384
mmcblk0p13      abootbackup               1024
mmcblk0p14      rpmbackup                  256
mmcblk0p15      utagsbackup                512
mmcblk0p16      tzbackup                   500
mmcblk0p17      versions                     4
mmcblk0p18      frp                        512
mmcblk0p19      mdm1m9kefs1               1024
mmcblk0p20      mdm1m9kefs2               1024
mmcblk0p21      mdm1m9kefs3               1024
mmcblk0p22      padb                      1024
mmcblk0p23      sp                       16384
mmcblk0p24      keystore                 16384
mmcblk0p25      logs                      2048
mmcblk0p26      persist                  32768
mmcblk0p27      mdm1hob                    256
mmcblk0p28      mdm1dhob                    32
mmcblk0p29      cid                        128
mmcblk0p30      logo                      8192
mmcblk0p31      misc                      1024
mmcblk0p32      padd                      2528
mmcblk0p33      mdm1m9kefsc                  1
mmcblk0p34      ssd                          8
mmcblk0p35      recovery                 16400
mmcblk0p36      kpan                      9088
mmcblk0p37      boot                     16384
mmcblk0p38      cache                   262144  /cache         ext4
mmcblk0p39      oem                      65536
mmcblk0p40      padc                      1024
mmcblk0p41      system                 2097152  /system        ext4
mmcblk0p42      userdata              27807616
                Total:                30500393

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  -------------------------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="media*" --exclude="*-cache"

Saving backup images in ./twrp-backup-2017-02-12--01-51-53/ ...
Saving partition boot (mmcblk0p37), 16 MiB uncompressed...
  boot.emmc.win: 100%   4.7 MiB/s   7.8 MiB
Saving tarball of mmcblk0p41 (mounted at /system), 2048 MiB uncompressed...
  system.ext4.win: 100%   5.7 MiB/s 834.3 MiB
Traceback (most recent call last):
  File "/usr/local/bin/tetherback", line 9, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "/usr/local/lib/python3.5/site-packages/tetherback/tetherback.py", line 338, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/usr/local/lib/python3.5/site-packages/tetherback/tetherback.py", line 229, in backup_partition
    raise RuntimeError("%s: don't know how to mount this partition" % pi.devname)
RuntimeError: mmcblk0p42: don't know how to mount this partition

```

Hi,

When trying to run your software (master or release), I get this error...
I'm running a Nexus 6, with CM installed.

If you need me to do any manipulations to help in finding the problem, please tell me and I will report back.

Thanks!

dlenski, 2017-02-12:
@cubox, thanks for including the partition map.

Can you run `adb shell cat /etc/fstab`  and post the output, while booted into TWRP? It seems that your `/data` partition may not be reported as such … perhaps because it's encrypted?

Cubox-, 2017-02-12:
`<cubox@Sleipnir>-<~> λ adb shell cat /etc/fstab
/dev/block/mmcblk0p41 /system ext4 rw 0 0
/dev/block/dm-0 /data ext4 rw 0 0
/dev/block/mmcblk0p38 /cache ext4 rw 0 0
/dev/block/mmcblk0p1 /firmware ext4 rw 0 0
 /usb-otg vfat rw 0 0`

If it's encrypted, it is with the default password. Android documentation reports that such password is literally "default_password"

TWRP is able to access /data through the "File Manager" option.

dlenski, 2017-02-14:
Basically the problem here is that [`build_partmap()`](https://github.com/dlenski/tetherback/blob/master/tetherback/tetherback.py#L127) is too naive in the way that it builds the map of block devices and partitions:

```
BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p42      userdata              27807616
```

What it does is first make a list of all the `mmcblk*` partitions, and then second it tries to figure out where those partitions are mounted.

Really, there should be more of a meet-in-the-middle approach:

* First, try to figure out what each block device partition's name means (e.g. `mmcblk0p42` is the `userdata` partition usually mounted at `/data`)
* Second, list all the mount points in `/etc/fstab` and try to figure out what they correspond to (e.g. `/data` is the mount point for `/dev/block/dm-0`)
* Third, realize that `mmcblk0p42` is the "raw" block device for the `userdata`/`/data/` partition (which will be used to make a raw partition image in the case of a Nandroid backup) while `/dev/block/dm-0` is the "mountable" block device (which will be mounted at `/data` in the case of a TWRP backup).

We'd end up with a partition table more like this:

```
BLOCK DEVICE                          PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
-----------------------------------   ----------------  ------------  -------------  --------
(raw) /sys/block/mmcblk0/mmcblk0p42   userdata              27807616  /data          ext4
(mountable) /dev/block/dm-0

(both) /sys/block/mmcblk0/mmcblk0p41  system                 2097152  /system        ext4

(both) /sys/block/mmcblk0/mmcblk0p1   modem                   114688  /firmware      ext4
```

I don't enough time to work on tetherback anymore… and I don't have a real device that would require this treatment so it'd be hard for me to test, anyway. Anyone care to take a crack at this? It should not be too hard to modify `build_partmap()` to be a bit more sophisticated and figure out the correct partition map as shown above.

Cubox-, 2017-02-14:
Thanks for your explanation on the issue.

I am not skilled to fix that, so I will leave this to someone who can.
However, is there a way for me to fix this issue temporarily (even by doing some ugly code/edits) for my device?

If I can hardcode the behaviour to follow into the code, or something of the sort.

Thanks!

nohajc, 2018-06-17:
Since my pull request has been merged, this can be closed.

closed 2018-06-18 by dlenski