dlenski/tetherback #45: Fails to backup on LG D802 (LG G2 International version)

Ghostbird, 2017-02-23:
I don't have logs available, but I thought I'd mention it here.
The resulting ext4 images are non-mountable after decompression, and restoring them with TWRP gave various interesting kinds of soft-bricks.

dlenski, 2017-02-28:
> The resulting ext4 images are non-mountable after decompression'

What do you mean by this?

If the file is named `.ext4.win`, it is *not* a raw image of an `ext4` partition, but it is actually a tarball of the contents of the mounted partition.

Hey, TWRP came up with that naming, not me. ¯\\\_(ツ)\_\/¯

Ghostbird, 2017-03-01:
What I meant by that, is that I decompressed the gzipped tarball and tried to mount the resulting ext4 filesystem, but it failed.

dlenski, 2017-03-01:
> Are you saying it's not an ext4 image at all in the gzipped tarball? 

Correct. It is just a tarball (`.tar.gz`) of the files on that filesystem, exactly that same as TWRP produces. If you want the raw image of the filesystem, then use `--nandroid` mode instead, as explained in the docs.

closed 2017-03-01 by dlenski

Ghostbird, 2017-03-01:
Ah, thanks, I misunderstood it. Regardless, the .ex4.win files could be restored with TWRP, but this did not result in a bootable system, so something was wrong. However, my attempts to find out what was wrong failed because I assumed the files were images.

dlenski, 2017-03-01:
> Regardless, the .ex4.win files could be restored with TWRP, but this did not result in a bootable system, so something was wrong.

Yes, I've had the same problem on `hammerhead` (Nexus5) unfortunately :-1:. Restoring the `/data` partition from a tarball appears to work 100% fine, but restoring the `/system` partition often does not.

My best guess is that something in the Android loader expects some file to be at a hard-coded location in the partition. Unfortunately I know very little about the Android boot process and don't have much time to learn about it these days.

Ghostbird, 2017-03-03:
No big deal, I just thought I'd mention it for the record. In my case even restoring `/data` didn't work.

dlenski, 2017-03-03:
Thanks, @Ghostbird. I should probably add a warning about the fact that it might not be possible to restore a functioning system from the TWRP-style backup images :-1:.

Do you happen to know if TWRP's built-in tarball backup *does* work to restore your `/data` partition correctly? I would love to figure out what metadata I am missing.

Ghostbird, 2017-03-03:
Sorry, I have no idea, I used tetherback because the device had insufficient storage for the backup, so I could not make a native TWRP backup.