dlenski/tetherback #50: ValueError: Value out of range, should be between 0 and 10322051072

ShapeShifter499, 2017-04-24:
I get the error as mentioned in the issue title and I'm not 100% sure why.  I do have a adopted storage sd card.  Device is a Moto X Pure with TWRP 3.1.0 and LinegaeOS 14.1 installed.

```
[raansu@Y40-80 tetherback]$ tetherback -N -M -D -R -C 
usage: tetherback [-h] [--version] [-s DEVICE_ID] [-o OUTPUT_PATH] [-N] [-0]
                  [-V] [-v] [-f] [-t | -x | -6 | -P] [-M] [-D] [-R] [-C] [-U]
                  [-E] [-S] [-B] [-X NAME] [--extra-raw NAME]
tetherback: error: -M/--media is deprecated. /data/media* will be included by default. Use -E/--no-media to exclude.
[raansu@Y40-80 tetherback]$ tetherback -N -D -R -C 
tetherback v0.9.1
Found ADB version 1.0.36
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.10.84-perf-g0ec7138
Device reports TWRP version 3.1.0-0
Reading partition map for mmcblk0 (48 partitions)...
  partition map: 100%                                                          
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                          
Reading partition map for mmcblk1 (2 partitions)...
  partition map: 100%                                                          
Saving backup images in ./nandroid-backup-2017-04-24--04-10-33/ ...
Saving partition boot (mmcblk0p36), 40 MiB uncompressed...
  boot.emmc.gz: 100%   5.1 MiB/s  10.5 MiB                                     
Saving partition recovery (mmcblk0p37), 40 MiB uncompressed...
  recovery.emmc.gz: 100%   6.3 MiB/s  14.6 MiB                                 
Saving partition system (mmcblk0p47), 4096 MiB uncompressed...
  system.emmc.gz: 100%   3.1 MiB/s 494.2 MiB                                   
Saving partition userdata (mmcblk0p48), 9843 MiB uncompressed...
  userdata.emmc.gz:  99%   8.3 MiB/s   9.6 GiB                                 Traceback (most recent call last):
  File "/usr/bin/tetherback", line 11, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "/usr/lib/python3.6/site-packages/tetherback/tetherback.py", line 344, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/usr/lib/python3.6/site-packages/tetherback/tetherback.py", line 284, in backup_partition
    pbar.update(out.tell())
  File "/usr/lib/python3.6/site-packages/progressbar/bar.py", line 501, in update
    % (self.min_value, self.max_value))
ValueError: Value out of range, should be between 0 and 10322051072
[raansu@Y40-80 tetherback]$ 
```

ShapeShifter499, 2017-04-24:
This is on Arch Linux with Python 3

dlenski, 2017-04-24:
This is a bug or corner case in the `progressbar` package which I've seen before… what version are you using?

```sh
$ pip3 list|grep progressbar2
progressbar2 (3.10.0)
```

ShapeShifter499, 2017-04-24:
progressbar2 (3.12.0)

ShapeShifter499, 2017-04-28:
@dlenski I have a problem related to the issue above.  I killed my phone doing something stupid but hooray Motorola has agreed to replace it and I'll be getting that today. The issue is that the TWRP backup of my old device did not include the decryption key for my adopted storage SD card which I know is kept at /data/misc/vold      I have the image file created after that error mentioned above and I cannot seem to read from it. 

After running `zcat userdata.emmc.gz > userdata.emmc`   I tried   `sudo mount userdata.emmc /mnt/f2fs -o loop -t f2fs`   with no luck

I also tried ext4 even though I'm pretty sure I had TWRP format my /data as f2fs

Is it still possible to mount that file somehow?  I'm hoping the adopted storage SD card key is in there.

ShapeShifter499, 2017-05-01:
@dlenski after some messing around I was able to fully restore my data.

dlenski, 2017-05-02:
Huh, what was wrong with it? Was the copy from `tetherback` corrupted?

This is, frankly, the only thing I really care about in the program… tetherback should never, never transfer a corrupt file (and if it does, it must complain loudly and fail immediately).

ShapeShifter499, 2017-05-02:
@dlenski   Your program copied the encrypted /data partition and I still don't know how to decrypt it. I know the original passphrase but there wasn't much information I could find on how to decrypt a image made from a encrypted /data partition.

 I did copy on my own a 'dd' image from /dev/block/dm-0  which is where TWRP mounted the decrypted data. I did something wrong as it copied bits from the terminal directly into the same image file but I did find the one file I needed to decrypt my adopted storage SD card.