dlenski/tetherback #51: Tetherback doesn't know how to mount a partition

Xalaxis, 2017-05-27:
Here's the log:
```C:\Users\matth\Desktop>tetherback -vP -SB
tetherback v0.9.1
Found ADB version 1.0.36
WARNING: adb shell pipe (--pipe) not needed with ADB >= 1.0.32; changing to adb exec-out pipe
Device reports kernel 3.10.84-perf+
Device reports TWRP version 3.1.0-0
Reading partition map for mmcblk0 (43 partitions)...
  partition map: 100%
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       modem                    81920  /firmware      vfat
mmcblk0p2       bluetooth                 1024
mmcblk0p3       pmic                       128
mmcblk0p4       pmicbak                    128
mmcblk0p5       limits                       1
mmcblk0p6       ddr                       1024
mmcblk0p7       sec                        128
mmcblk0p8       fsg                       1536
mmcblk0p9       devinfo                      1
mmcblk0p10      aboot                     1024
mmcblk0p11      abootbak                  1024
mmcblk0p12      apdp                       256
mmcblk0p13      msadp                      256
mmcblk0p14      dpo                          1
mmcblk0p15      fsc                          1
mmcblk0p16      ssd                          8
mmcblk0p17      modemst1                  1536
mmcblk0p18      modemst2                  1536
mmcblk0p19      oem_dycnvbk              10240
mmcblk0p20      oem_stanvbk              10240
mmcblk0p21      param                     1024
mmcblk0p22      reserve1                  8096
mmcblk0p23      reserve2                 16192
mmcblk0p24      logo                     16384
mmcblk0p25      driver                   32768
mmcblk0p26      sbl1                      1024
mmcblk0p27      sbl1bak                   1024
mmcblk0p28      sdi                         96
mmcblk0p29      tz                        1024
mmcblk0p30      tzbak                     1024
mmcblk0p31      rpm                        500
mmcblk0p32      rpmbak                     500
mmcblk0p33      hyp                        512
mmcblk0p34      hypbak                     512
mmcblk0p35      boot                     65536
mmcblk0p36      recovery                 65536
mmcblk0p37      persist                  32768
mmcblk0p38      misc                      1024
mmcblk0p39      keystore                   512
mmcblk0p40      config                     512
mmcblk0p41      cache                   262144  /cache         ext4
mmcblk0p42      system                 2621440  /system        ext4
mmcblk0p43      userdata              57663471
                Total:                60905635

Backup plan:

PARTITION NAME    FILENAME       FORMAT
----------------  -------------  ------------------------------
userdata          data.ext4.win  tar -cz -p --exclude="*-cache"

Saving backup images in .\twrp-backup-2017-05-26--18-59-06/ ...
Traceback (most recent call last):
  File "C:\Users\matth\AppData\Local\Programs\Python\Python36\Scripts\tetherback-script.py", line 11, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "c:\users\matth\appdata\local\programs\python\python36\lib\site-packages\tetherback\tetherback.py", line 344, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "c:\users\matth\appdata\local\programs\python\python36\lib\site-packages\tetherback\tetherback.py", line 230, in backup_partition
    raise RuntimeError("%s: don't know how to mount this partition" % pi.devname)
RuntimeError: mmcblk0p43: don't know how to mount this partition```

I'm thinking this might be related to this being an encrypted device?  Once TWRP has decrypted the /data partition does Tetherback know where to look for the contents?

Xalaxis, 2017-05-27:
This is the Oneplus 2

dlenski, 2017-05-27:
Yes, this is due to encryption. The `userdata` partition can't be mounted directly because of it.

If you want to back it up while encrypted, you'll need to do a "Nandroid style" backup which just saves an image of the raw partition, rather than its contents.

I don't own any encrypted device, so I haven't been able to work on improving this.

Xalaxis, 2017-05-28:
Thanks, I'm doing a Nandroid-style backup now.  Do you really not have any Android device that supports encryption?  Support for it goes all the way back to ICS (4.0+)

Xalaxis, 2017-05-28:
As a triage measure, is there a chance we can add dedicated detection for encrypted partitions or simply add '(This may be because this partition is encrypted. If this is the case please perform a Nandroid-style backup)' to the error message?

dlenski, 2017-05-28:
Pull requests for any of these changes are welcome. :grinning:

I have become fairly pessimistic about the ability to make this tool work stably across many devices and Android versions. And the TWRP recovery project itself is adding support for backup over USB, which I think will be a better way to support this long-term …

Xalaxis, 2017-06-02:
If TWRP is adding native support then this sort of makes this app redundant. Thank you for your initial development work however, this app has saved me more than a few times 😄

closed 2017-06-02 by Xalaxis