dlenski/tetherback #53: Add the "nuclear option" to back up ALL partitions?

magicgoose, 2017-09-09:
…or there are reasons why this won't be useful?

dlenski, 2017-09-10:
It might be for someone somewhere who's hacking on Android firmware… but the `/boot`, `/data`, `/system` partitions *should* be enough to recover (almost?) any Android system from a totally wiped state.

Pull requests are welcome if you'd like to add an `--all-partitions` option or something to that effect :-D

magicgoose, 2017-09-10:
Well, it depends what do we mean by totally wiped state — AFAIK modem firmware can easily get borked if one tries to do ambitious things with it, for example. Also, in theory any partition may get damaged data due to "bit rot" eventually…

I'll probably make a PR if/when I have time, but can't give a promise yet.

dlenski, 2017-09-10:
> AFAIK modem firmware can easily get borked if one tries to do ambitious things with it, for example

Right. That's one reason why you might want to backup the `modem*` partitions—but also a reason why you might _not_ want to back them up.

So I'd be hesitant to tell anyone to restore these partitions without knowing what they're doing!