dlenski/tetherback #54: RuntimeError: mmcblk0p28: don't know how to mount this partition

Krutonium, 2017-10-24:
```
[krutonium@krutonium-msi ~]$ tetherback
tetherback v0.9.1
Found ADB version 1.0.39
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.4.0-bricked-hammerhead-twrp-g7b77eb4
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100%                                                                                                                                                                        
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                                                                                                                                        
Saving backup images in ./twrp-backup-2017-10-23--22-20-15/ ...
Saving partition boot (mmcblk0p19), 22 MiB uncompressed...
  boot.emmc.win: 100%   2.9 MiB/s   8.5 MiB                                                                                                                                                  
Saving tarball of mmcblk0p25 (mounted at /system), 1024 MiB uncompressed...
  system.ext4.win: 100%   2.6 MiB/s 437.4 MiB                                                                                                                                                
Traceback (most recent call last):
  File "/usr/bin/tetherback", line 11, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "/usr/lib/python3.6/site-packages/tetherback/tetherback.py", line 344, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/usr/lib/python3.6/site-packages/tetherback/tetherback.py", line 230, in backup_partition
    raise RuntimeError("%s: don't know how to mount this partition" % pi.devname)
RuntimeError: mmcblk0p28: don't know how to mount this partition
```

Nexus 5 on CyanogenMod, Android 6.

dlenski, 2017-10-26:
What does `tetherback -0` show you for your partition map?

Krutonium, 2017-10-26:
```
[krutonium@krutonium-msi ~]$ tetherback -0
tetherback v0.9.1
Found ADB version 1.0.39
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel * daemon not running. starting it now at tcp:5037 *
* daemon started successfully *
3.4.0-bricked-hammerhead-twrp-g7b77eb4
Device reports TWRP version 3.0.2-0
Reading partition map for mmcblk0 (29 partitions)...
  partition map: 100%                                                                                                                                                                        
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                                                                                                                                        

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       modem                    65536  /firmware      vfat
mmcblk0p2       sbl1                      1024
mmcblk0p3       rpm                        512
mmcblk0p4       tz                         512
mmcblk0p5       sdi                        512
mmcblk0p6       aboot                      512
mmcblk0p7       pad                       2048
mmcblk0p8       sbl1b                     1024
mmcblk0p9       tzb                        512
mmcblk0p10      rpmb                       512
mmcblk0p11      abootb                     512
mmcblk0p12      modemst1                  3072
mmcblk0p13      modemst2                  3072
mmcblk0p14      metadata                   512
mmcblk0p15      misc                     16384
mmcblk0p16      persist                  16384  /persist       ext4
mmcblk0p17      imgdata                   3072
mmcblk0p18      laf                      22528
mmcblk0p19      boot                     22528
mmcblk0p20      recovery                 22528
mmcblk0p21      fsg                       3072
mmcblk0p22      fsc                        512
mmcblk0p23      ssd                        512
mmcblk0p24      ddr                        512
mmcblk0p25      system                 1048576  /system        ext4
mmcblk0p26      crypto                   30720
mmcblk0p27      cache                   716800  /cache         ext4
mmcblk0p28      userdata              13404138
mmcblk0p29      grow                         5
                Total:                15388143

Backup plan:

PARTITION NAME    FILENAME         FORMAT
----------------  ---------------  ------------------------------
boot              boot.emmc.win    gzipped raw image
system            system.ext4.win  tar -cz -p
userdata          data.ext4.win    tar -cz -p --exclude="*-cache"
```

Krutonium, 2017-10-26:
I just realised I said CyanogenMod, I meant to say LineageOS. That change still gets me.

dlenski, 2017-10-30:
Looks like you have an encrypted `/data` partition. `tetherback` doesn't have access to the keys so can't mount it, and therefore can't back it up in `--twrp` mode (but it _can_ copy the filesystem as a raw partition image in `--nandroid` mode).

There are so many different flavors of encryption that I can't keep up with 'em, and neither can the TWRP developers it seems :slightly_frowning_face:

closed 2017-10-30 by dlenski

Krutonium, 2017-10-30:
But it's already mounted via TWRP, and it also fails in Nandroid Mode...

dlenski, 2017-10-30:
> But it's already mounted via TWRP…

Ah, yes I overlooked that, thanks. I believe I've seen this case before.

The problem is that the underlying block device is `/dev/block/mmcblk0p16`, but the device that actually gets mounted as `/data` is the device-mapper encrypted device `/dev/block/dm-0`.

So tetherback:

1. can't figure out if/where it's mounted and thus can't do the tarball backup right
2. can't figure out which raw block device is associated with `userdata` and verify that it's *unmounted* in order to do the Nandroid backup. (Not 100% sure about this… maybe you can confirm.)

I don't have an encrypted device with which to play around and test this, and have more-or-less given up on supporting all the possible partition-mapping combinations due to the difficulty of me testing them.

If you can come up with a patch that can build the partition table and include both the "raw" `dd`-able partition and _and_ the `mount`-able partition, that would be very welcome.