dlenski/tetherback #65: HTC10 (pme) partition `/boot` not mounted

ckrmml, 2018-11-28:
When tetherback tries to backup the partitions, it tries to unmount them before backing them up. `/boot` partition is not mounted which crashes tetherback when it's trying to unmount the partition.  
Checking if a partition is really mounted before trying to unmount it solved the issue for me

Log of error:
```
(env) [chris@tp-w520 install]$ tetherback -N
tetherback v0.9.1
Found ADB version 1.0.40
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.18.31-perf-gdad85f6-dirty
Device reports TWRP version 3.2.0-0
Reading partition map for mmcblk0 (66 partitions)...
  partition map: 100%                                                                                                                                                                                          
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                                                                                                                                                          
Reading partition map for mmcblk1 (1 partitions)...
  partition map: N/A%                                                                                                                                                                                          WARNING: partition mmcblk1p1 has no PARTNAME in its uevent file
Please report this issue at https://github.com/dlenski/tetherback/issues
Please post the entire output from tetherback!
  partition map: 100%                                                                                                                                                                                          
Saving backup images in ./nandroid-backup-2018-11-28--12-56-04/ ...
Saving partition boot (mmcblk0p59), 64 MiB uncompressed...
Traceback (most recent call last):
  File "/home/shared/dev/android/tetherback/install/env/bin/tetherback", line 11, in <module>
    sys.exit(main())
  File "/home/shared/dev/android/tetherback/install/env/lib/python3.7/site-packages/tetherback/tetherback.py", line 359, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/home/shared/dev/android/tetherback/install/env/lib/python3.7/site-packages/tetherback/tetherback.py", line 255, in backup_partition
    if not really_umount(adb, '/dev/block/'+pi.devname, pi.mountpoint):
  File "/home/shared/dev/android/tetherback/install/env/lib/python3.7/site-packages/tetherback/adb_stuff.py", line 32, in really_umount
    if adb.check_output(('shell','umount %s 2>/dev/null && echo ok' % dev)).strip():
  File "/home/shared/dev/android/tetherback/install/env/lib/python3.7/site-packages/tetherback/adb_wrapper.py", line 35, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
  File "/usr/lib64/python3.7/subprocess.py", line 389, in check_output
    **kwargs).stdout
  File "/usr/lib64/python3.7/subprocess.py", line 481, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '('adb', '-d', 'shell', 'umount /dev/block/mmcblk0p59 2>/dev/null && echo ok')' returned non-zero exit status 1.
```

output of `adb shell mount`:
```
~ # mount
rootfs on / type rootfs (rw,seclabel,size=1757884k,nr_inodes=439471)
tmpfs on /dev type tmpfs (rw,seclabel,nosuid,relatime,mode=755)
devpts on /dev/pts type devpts (rw,seclabel,relatime,mode=600)
proc on /proc type proc (rw,relatime,gid=3009,hidepid=2)
sysfs on /sys type sysfs (rw,seclabel,relatime)
selinuxfs on /sys/fs/selinux type selinuxfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,seclabel,relatime)
pstore on /sys/fs/pstore type pstore (rw,seclabel,relatime)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
/dev/block/mmcblk0p61 on /cache type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/dm-0 on /data type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/dm-0 on /sdcard type ext4 (rw,seclabel,relatime,data=ordered)
/dev/block/mmcblk1p1 on /external_sd type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

dlenski, 2018-11-29:
Hrmmm… I wonder if this is related to #49.

> Checking if a partition is really mounted before trying to unmount it solved the issue for me

Do you have a patch to do this?

ckrmml, 2018-11-29:
This was the second issue in #58, i just split it up. I'm not sure if this is related to #49 but i will check that. 
  > Do you have a patch to do this?

  Currently i'm checking if the partition is mounted

      def is_mounted(adb, dev, node):   
          for l in adb.check_output(('shell','mount')).splitlines():  
              f = l.split()  
              if dev in f[0] and node in f[2]:  
                  return True  
          else:  
              return False  
  

  right before trying to unmount it in `really_umount`

  
      def really_umount(adb, dev, node):
          if is_mounted(adb, dev, node):
              for opts in ('','-f','-l','-r'):
                  if adb.check_output(('shell','umount %s 2>/dev/null && echo ok' % dev)).strip():
                      break
                  if adb.c[...]
  
But since i have got new errors (md5sum mismatches) i have to clear those out before i request a pull. 

When this is done i will try to solve #57 (at least for me, the partition handling seems to differ between Samsung and Huawei and i have no other devices with sd* partitions to test it). That should also solve parts of #48 and #39 (recognizing sd* scheme instead of mmcblk)