dlenski/tetherback #68: KeyError 'system' while executing tetherback

HeapUnderfl0w, 2019-03-20:
While executing tetherback i get the following error:
```
tetherback v0.9.1
Found ADB version 1.0.40
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.18.71-perf+
Device reports TWRP version 3.2.3
Reading partition map for mmcblk0 (2 partitions)...
  partition map: 100%                                                                                                                                                                                                                                                                                                        
Traceback (most recent call last):
  File "/home/byte/Documents/phone/nand/env/bin/tetherback", line 10, in <module>
    sys.exit(main())
  File "/home/byte/Documents/phone/nand/env/lib/python3.7/site-packages/tetherback/tetherback.py", line 338, in main
    plan = plan_backup(args, partmap)
  File "/home/byte/Documents/phone/nand/env/lib/python3.7/site-packages/tetherback/tetherback.py", line 206, in plan_backup
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
  File "/home/byte/Documents/phone/nand/env/lib/python3.7/site-packages/tetherback/tetherback.py", line 206, in <genexpr>
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
KeyError: 'system'
```
The phone is in recovery and recognized by ADB, so i dont know what could be causing this.
I tried both the release and latest version.

PC OS: ArchLinux `Linux sigma 5.0.2-arch1-1-ARCH #1 SMP PREEMPT Thu Mar 14 18:47:49 UTC 2019 x86_64 GNU/Linux`
Phone: LG-G6 (H870) using  [HavocOS Unofficial](https://forum.xda-developers.com/lg-g6/development/rom-havocos-v2-1-t3893668)
Python Version: `Python 3.7.2`
Pip Version: `pip 19.0.3 from /home/byte/Documents/phone/nand/env/lib/python3.7/site-packages/pip (python 3.7)`
I am using a virtualenv

dlenski, 2019-07-29:
What does `tetherback -0v` show for the partition map?

HeapUnderfl0w, 2019-08-03:
nothing, it crashes after retrieving it:

```
tetherback v0.9.1
Found ADB version 1.0.40
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.18.71-perf+
Device reports TWRP version 3.2.3
Reading partition map for mmcblk0 (2 partitions)...
  partition map: 100%                                                                                                                                                                                                                                                                                                        
Traceback (most recent call last):
  File "/home/byte/.local/bin/tetherback", line 11, in <module>
    load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')()
  File "/home/byte/.local/lib/python3.7/site-packages/tetherback/tetherback.py", line 338, in main
    plan = plan_backup(args, partmap)
  File "/home/byte/.local/lib/python3.7/site-packages/tetherback/tetherback.py", line 206, in plan_backup
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
  File "/home/byte/.local/lib/python3.7/site-packages/tetherback/tetherback.py", line 206, in <genexpr>
    plan.update((p,BackupPlan('%s.%s.win'%(p, partmap[p].fstype), '-p')) for p in mp)
KeyError: 'system'
```
i have updated to the latest dev release and it is still happening.
a note that i forgot to add, due to the nature of the LG-G6 it is using a patched version of TWRP to allow proper access to patched partitions (release name is: `3.2.3-Nebula_Alpha_20180813`). I assumed this would not prove a problem (as the interface and outputs should be unchanged).