dlenski/tetherback #77: Error: Python 3.6 or newer is required

JonnyTech, 2023-08-13:
My version of python is definitely > 3.6:

```
$ python --version
Python 3.11.2

$ python3 --version
Python 3.11.2

$ python setup.py 
Python 3.6 or newer is required.

$ python3 setup.py 
Python 3.6 or newer is required.

$ python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.version_info
sys.version_info(major=3, minor=11, micro=2, releaselevel='final', serial=0)

$ pip3 install https://github.com/dlenski/tetherback/archive/HEAD.zip
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      Python 3.6 or newer is required.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Cannot determine package name from spec 'https://github.com/dlenski/tetherback/archive/HEAD.zip'.
Check package spec for errors.

$ head -n 8 setup.py 
#!/usr/bin/env python3

from __future__ import print_function
import sys, os, re, subprocess as sp
from setuptools import setup

if not sys.version_info < (3, 6):
    sys.exit("Python 3.6 or newer is required.")

```

dreirund, 2023-08-19:
I have this issue, too; trying to install latest git checkout.

dreirund, 2023-08-19:
I have this issue, too; trying to install latest git checkout.

Python version: 3.11.3.

`setup.py` has  
```
if not sys.version_info < (3, 6):
    sys.exit("Python 3.6 or newer is required.")
```

This will abort for any python version >= 3.6. I think the `not` should not be there.

Please fix.

Regards!

dreirund, 2023-08-19:
I have this issue, too; trying to install latest git checkout.

Python version: 3.11.3.

`setup.py` has  
```
if not sys.version_info < (3, 6):
    sys.exit("Python 3.6 or newer is required.")
```

This will abort for any python version >= 3.6. I think the `not` should not be there.

I have attached a patch that fixes this:  
[`setup.py-fix-version-detection.patch.txt`](https://github.com/dlenski/tetherback/files/12385639/setup.py-fix-version-detection.patch.txt)

Regards!

chr4ss12, 2023-10-30:
download HEAD.zip from this repo (instructions in readme.md), unpack, edit setup.py manually to remove the check, and pack it up again to HEAD.zip, and install it locally using pip3 install HEAD.zip, worked for me, this quick hack.

dreirund, 2023-10-30:
> *download HEAD.zip from this repo (instructions in readme.md), unpack, edit setup.py manually to remove the check, and pack it up again to HEAD.zip, and install it locally using pip3 install HEAD.zip, worked for me, this quick hack.*

This does _not_ fix the underlying issue reported here. The fix should be implemented in the code hosted in this repository.

chr4ss12, 2023-10-31:
you could make a PR.

dreirund, 2023-10-31:
> you could make a PR.

That is beyond my git skills.

If a fix is provided here to the maintainers of the software, than it
should be easy to just implement it by people who have write
permissions to the repository anyway.

dlenski, 2024-06-07:
> >  you could make a PR.
> That is beyond my git skills.

…?

If you know how to create a patch, as in https://github.com/dlenski/tetherback/issues/77#issuecomment-1684918341, then you can definitely handle making a PR.

closed 2024-06-07 by dlenski

dreirund, 2024-06-07:
> You appear to have opened _hundreds_ of Github issues in the past couple years, @dreirund, and since you know how to create patches, as in [#77 (comment)](https://github.com/dlenski/tetherback/issues/77#issuecomment-1684918341), you can _definitely_ handle making a PR.

Two things to this:

1. I do not know any more git than `git clone`, `git add`, `git commit`, `git push` (just enough to get source code, and _very basic_ upload to repositories I have write access to; which in most cases it the AUR package recipe. If anythings goes wrong there I delete my local repo and re-clone from the AUR, because I fail to revert changes.)  
   I know more advanced git usage _can_ all be learned, but I haven't.
2. I see it as a service to projects to submit patches (if I have a patch I have already fixed it for myself, so I do not need the fix anymore, so reporting the issue _with_ mentioning a fix is a pure service for upstream). So I also do not see any moral obligation to do the more work to create a merge request/ pull request/ or however it is called, when I already created a patch that works for me.

(Amd 3., I think quite a lot of the "_hundrets_" of issues are because of the website reporting tool of a cookie banner addon I use; reporting in that tool automatically opens an issue at github.)

Regards and thanks for the maintanance of the software!

dreirund, 2024-06-07:
> > You appear to have opened _hundreds_ of Github issues in the past couple years, @dreirund, and since you know how to create patches, as in [#77 (comment)](https://github.com/dlenski/tetherback/issues/77#issuecomment-1684918341), you can _definitely_ handle making a PR.
> 
> Two things to this:
> 
>     1. I do not know any more git than `git clone`, `git pull`, `git add`, `git commit`, `git push` (and some checkout of a branch, though this sometimes also produces unexpected results, then I delete and restart from scratch) (just enough to get source code, and _very basic_ upload to repositories I have write access to; which in most cases is the [AUR](http://aur.archlinux.com/) package recipe. If anythings goes wrong there I delete my local repo and re-clone from the AUR, because I fail to revert changes; and the AUR also initialises the repo for me).
>        I know more advanced git usage _can_ all be learned, but I haven't.
> 
>     2. I see it as a service to projects to submit patches (if I have a patch I have already fixed it for myself, so I do not need the fix anymore, so reporting the issue _with_ mentioning a fix is a pure service for upstream). So I also do not see any moral obligation to do the more work to create a merge request/ pull request/ or however it is called, when I already created a patch that works for me.
> 
> 
> (And 3., I think quite a lot of the "_hundrets_" of issues are because of the website reporting tool of a cookie banner addon I use; reporting in that tool automatically opens an issue at github.)
> 
> Regards and thanks for the maintanance of the software!

commit e5394152 (2024-06-07): Fix version check

Fixes #77, which was caused by my slapdash change in
1373ca67e43e6b95e24e7893daf4649f8d3eedf4.

Thanks @dreirund for the patch in
https://github.com/dlenski/tetherback/issues/77#issuecomment-1684918341.