dlenski/tetherback #79: Do not try to unmount non-mounted particions.

dreirund, 2023-08-19:
I have [tetherbacl version 0.9.1](https://github.com/dlenski/tetherback/releases/tag/0.9.1) installed.

It tries to unmount partitions that are not mounted, which results in an error.

The script should not unmount what is not mounted, or handle unmount-failures in case the partition is not mounted without error:

`tetherback -v -M -D -R -C`:  
```
tetherback v0.9.1
Found ADB version 1.0.41
Using default transfer method: adb exec-out pipe (--exec-out)
Device reports kernel 3.18.140-Perf+_r5.6
Device reports TWRP version 3.5.2
Reading partition map for mmcblk0 (49 partitions)...
  partition map: 100%                                                                                                                                       
Reading partition map for mmcblk0rpmb (0 partitions)...
  partition map: 100%                                                                                                                                       
Reading partition map for mmcblk1 (2 partitions)...
  partition map: 100%                                                                                                                                       

Partition map:

BLOCK DEVICE    PARTITION NAME      SIZE (KiB)  MOUNT POINT    FSTYPE
--------------  ----------------  ------------  -------------  --------
mmcblk0p1       modem                    86016  /firmware      vfat
mmcblk0p2       fsc                          1
mmcblk0p3       ssd                          8
mmcblk0p4       sbl1                       512
mmcblk0p5       sbl1bak                    512
mmcblk0p6       rpm                        512
mmcblk0p7       rpmbak                     512
mmcblk0p8       tz                        2048
mmcblk0p9       tzbak                     2048
mmcblk0p10      devcfg                     256
mmcblk0p11      devcfgbak                  256
mmcblk0p12      dsp                      16384
mmcblk0p13      modemst1                  1536
mmcblk0p14      modemst2                  1536
mmcblk0p15      ddr                         32
mmcblk0p16      fsg                       1536
mmcblk0p17      sec                         16
mmcblk0p18      splash                   20480
mmcblk0p19      aboot                     1024
mmcblk0p20      abootbak                  1024
mmcblk0p21      boot                     65536
mmcblk0p22      recovery                 65536
mmcblk0p23      devinfo                   1024
mmcblk0p24      system                 3145728  /system_root   ext4
mmcblk0p25      cache                   262144  /cache         f2fs
mmcblk0p26      persist                  32768  /persist       ext4
mmcblk0p27      misc                      1024
mmcblk0p28      keystore                   512
mmcblk0p29      config                      32
mmcblk0p30      oem                      65536
mmcblk0p31      limits                      32
mmcblk0p32      mota                       512
mmcblk0p33      dip                       1024
mmcblk0p34      mdtp                     32768
mmcblk0p35      syscfg                     512
mmcblk0p36      mcfg                      4096
mmcblk0p37      lksecapp                   128
mmcblk0p38      lksecappbak                128
mmcblk0p39      cmnlib                     256
mmcblk0p40      cmnlibbak                  256
mmcblk0p41      cmnlib64                   256
mmcblk0p42      cmnlib64bak                256
mmcblk0p43      keymaster                  256
mmcblk0p44      keymasterbak               256
mmcblk0p45      apdp                       256
mmcblk0p46      msadp                      256
mmcblk0p47      dpo                          8
mmcblk0p48      cust                    524288  /vendor        ext4
mmcblk0p49      userdata              25685479
mmcblk1p1       android_meta             16384  /sdcard1       auto
mmcblk1p2       android_expand       124851183
                Total:               154894679

Backup plan:

PARTITION NAME    FILENAME           FORMAT
----------------  -----------------  -----------------
boot              boot.emmc.win      gzipped raw image
recovery          recovery.emmc.win  gzipped raw image
cache             cache.f2fs.win     tar -cz -p
system            system.ext4.win    tar -cz -p
userdata          data.ext4.win      tar -cz -p

Saving backup images in ./twrp-backup-2023-08-19--12-38-24/ ...
Saving partition boot (mmcblk0p21), 64 MiB uncompressed...
Traceback (most recent call last):
  File "/usr/bin/tetherback", line 33, in <module>
    sys.exit(load_entry_point('tetherback==0.9.1', 'console_scripts', 'tetherback')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/tetherback/tetherback.py", line 338, in main
    backup_partition(adb, partmap[standard], bp, args.transport, backupdir, args.verify)
  File "/usr/lib/python3.11/site-packages/tetherback/tetherback.py", line 239, in backup_partition
    if not really_umount(adb, '/dev/block/'+pi.devname, pi.mountpoint):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/tetherback/adb_stuff.py", line 32, in really_umount
    if adb.check_output(('shell','umount %s 2>/dev/null && echo ok' % dev)).strip():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/tetherback/adb_wrapper.py", line 35, in check_output
    return sp.check_output(self.adbcmd(adbargs), universal_newlines=un, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '('adb', '-d', 'shell', 'umount /dev/block/mmcblk0p21 2>/dev/null && echo ok')' returned non-zero exit status 1.
```

dreirund, 2023-08-19:
... seems to be fixed in latest git version; please do a release which has the fix. Your latest release is many years old now.

ossilator, 2024-05-28:
duplicate of #65.